The-OpenROAD-Project/OpenROAD #5284: rudy covers only 1/4 of the diearea

stefanottili, 2024-06-25:
### Describe the bug

the Rudy map looks correct, but is displayed in the lower quadrant of the diearea 

UNITS are 1000
DIEAREA ( -479600 -479600 ) ( 480000 480000 ) ;

[image]

### Expected Behavior

Rudy map should match the diearea

### Environment

```shell
M1 MacOS
OpenROAD v2.0-14310-g69872cd76 (Jun 25 2024)
Features included (+) or not (-): +Charts +GPU +GUI +Python
```

### To Reproduce

openroad -gui -threads max risc2.or

read_lef lef/risc2.lef.gz
read_def def/risc2.def.gz
global_placement
detailed_placement

[iccad04risc2.tgz](https://github.com/user-attachments/files/15975330/iccad04risc2.tgz)

stefanottili, 2024-06-25:
Unrelated to Rudy, but using the same lef/def.
Let's try mpl2 to get at better macro placement: 

>>> macro_placement
[WARNING MPL-0100] No macros found.

The MACRO's are PLACED and of CLASS RING.

AcKoucher, 2024-06-25:
FYI @stefanottili This warning in from the original macro placer mpl, not mpl2.

stefanottili, 2024-06-25:
changing CLASS RING to CLASS BLOCK get's me a MACRO placement, but then detailed placement fails with
```
NesterovSolve] Iter:  420 overflow: 0.105 HPWL: 1231953979
[NesterovSolve] Finished with Overflow: 0.099651
[WARNING DPL-0038] No 1-site fill cells detected.  To remove 1-site gaps use the -disallow_one_site_gaps flag.
[INFO DPL-0034] Detailed placement failed on the following 62 instances:
[INFO DPL-0035]  lx1/lx0/IRAM/MUX_INSTW_UP/U27
...
[ERROR DPL-0036] Detailed placement failed.
```
I'm building OpenRoad with mpl2, so why wouldn't it be used ?

[ 74%] Built target mpl2_lib
[ 74%] Built target mpl2

[image]

```
read_lef lef/risc2.lef.gz
read_def def/risc2.def.gz
macro_placement
global_placement -density 0.95
detailed_placement
```

stefanottili, 2024-06-25:
> Am I corred to assume that mpl2 won't derive hierarchy information from def using the DIVIDERCHAR info ?
> So I need to have hierarchical verilog to run rtl_macro_placer ?

jupp, confirmed, rtl_macro_placer coredumps when just reading lef/def.

Side notes: 
a) why not rtl_macro_placement as in macro_placement/global_placement/detailed_placement ?
b) why not hier_macro_placement as there is no rtl involved, just gate level netlist hierarchy ?
```
read_lef lef/risc2.lef.gz
read_def def/risc2.def.gz
rtl_macro_placer
```
```
[INFO ODB-0227] LEF file: lef/risc2.lef.gz, created 13 layers, 16 vias, 147 library cells
[INFO ODB-0127] Reading DEF file: def/risc2.def.gz
[INFO ODB-0128] Design: frisc
[INFO ODB-0130]     Created 629 pins.
[INFO ODB-0131]     Created 32622 components and 196041 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 65244 connections.
[INFO ODB-0133]     Created 34034 nets and 128506 connections.
[INFO ODB-0134] Finished DEF file: def/risc2.def.gz
Floorplan Outline: (-479.6, -479.6) (480, 480),  Core Outline: (-479.6, -478.4) (480, 479.2)
Traversed logical hierarchy
	Number of std cell instances: 32615
	Area of std cell instances: 521777.56
	Number of macros: 7
	Area of macros: 349004.25
	Area of macros with halos: 349004.25
	Area of std cell instances + Area of macros: 870781.81
	Core area: 918912.94
	Design Utilization: 0.95
	Core Utilization: 0.92
	Manufacturing Grid: 10

Signal 11 received
Stack trace:
 0# handler(int) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# _sigtramp in /usr/lib/system/libsystem_platform.dylib
 2# mpl2::HierRTLMP::createDataFlow() in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# mpl2::HierRTLMP::createDataFlow() in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# mpl2::HierRTLMP::runMultilevelAutoclustering() in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# mpl2::HierRTLMP::run() in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# mpl2::MacroPlacer2::place(int, int, int, int, int, float, int, float, int, int, int, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, int, bool, char const*) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# mpl2::rtl_macro_placer_cmd(int, int, int, int, float, int, float, int, int, int, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, int, bool, char const*) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# _wrap_rtl_macro_placer_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# TclNRRunCallbacks in /opt/homebrew/Cellar/tcl-tk/8.6.14/lib/libtcl8.6.dylib
10# TclEvalEx in /opt/homebrew/Cellar/tcl-tk/8.6.14/lib/libtcl8.6.dylib
11# Tcl_Eval in /opt/homebrew/Cellar/tcl-tk/8.6.14/lib/libtcl8.6.dylib
12# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# gui::startGui(int&, char**, Tcl_Interp*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, bool) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# ord::tclAppInit(Tcl_Interp*) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# Tcl_MainEx in /opt/homebrew/Cellar/tcl-tk/8.6.14/lib/libtcl8.6.dylib
16# TclInitNamespaceSubsystem in /opt/homebrew/Cellar/tcl-tk/8.6.14/lib/libtcl8.6.dylib
17# main in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
```

eder-matheus, 2024-06-25:
@stefanottili Could you create a new issue reporting the MPL2 crash? Discussing MPL2 in the rudy issue might the discussion very confusing here.

eder-matheus, 2024-06-27:
@stefanottili, I am testing your design with the latest master branch and no longer see the issue. I believe PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/5293 fixed it:
[image]

 I will close this now, but please feel free to open it again if the error happens again.

closed 2024-06-27 by eder-matheus