The-OpenROAD-Project/OpenROAD #5286: [ERROR GRT-0076] Net T_ICK not properly covered.

stefanottili, 2024-06-25:
### Describe the bug

global route fails.

Is OpenRoad supposed to be able to handle centered DIEAREA ?

DIEAREA ( -321200 -321200 ) ( 321600 321600 ) ;

### Expected Behavior

global route should finish

### Environment

```shell
M1 MacOS 
OpenROAD v2.0-14310-g69872cd76  (Jun 25 2024)
Features included (+) or not (-): +Charts +GPU +GUI +Python
```

### To Reproduce

[iccad04_dsp2.tgz](https://github.com/user-attachments/files/15978987/iccad04_dsp2.tgz)

### Relevant log output

```shell
OpenROAD v2.0-14310-g69872cd76 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 8 thread(s).
[WARNING ODB-0220] WARNING (LEFPARS-2003): BUSBITCHARS is a required statement on LEF file with version 5.5 and earlier.
Without BUSBITCHARS defined, the LEF file is technically incorrect.
Refer the LEF/DEF 5.5 or earlier Language Referece manual on how to define this statement. See file lef/dsp2.lef.gz at line 34674.

[WARNING ODB-0220] WARNING (LEFPARS-2004): DIVIDERCHAR is a required statementon LEF file with version 5.5 and earlier.
Without DIVIDECHAR defined, the LEF file is technically incorrect.
Refer the LEF/DEF 5.5 or earlier Language Referece manual on how to define this statement. See file lef/dsp2.lef.gz at line 34674.

[INFO ODB-0227] LEF file: lef/dsp2.lef.gz, created 13 layers, 16 vias, 293 library cells
[INFO ODB-0127] Reading DEF file: def/dsp2.def.gz
[INFO ODB-0128] Design: DSP_CORE
[INFO ODB-0130]     Created 846 pins.
[INFO ODB-0131]     Created 26281 components and 143857 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 52562 connections.
[INFO ODB-0133]     Created 28431 nets and 90973 connections.
[INFO ODB-0134] Finished DEF file: def/dsp2.def.gz
[WARNING GRT-0300] Timing is not available, setting critical nets percentage to 0.
[ERROR GRT-0076] Net T_ICK not properly covered.
[ERROR GUI-0070] Error: dsp2.or, 3 GRT-0076
```

stefanottili, 2024-06-26:
> Net T_ICK not properly covered

What is this error message supposed to convey ? 
Not properly covered by what ?

eder-matheus, 2024-06-26:
> > Net T_ICK not properly covered
> 
> What is this error message supposed to convey ? Not properly covered by what ?

It's a bad message. The error happens when a net does not have global routing, but it was expected to have at this point. I updated the message in my PR with the fix.

closed 2024-06-27 by eder-matheus