The-OpenROAD-Project/OpenROAD #5290: Multiple warnings with ihp-sg13g2/aes on master

gudeh, 2024-06-26:
### Describe the bug

This warning is surpassing the maximum allowed: `[WARNING GRT-0041] Net VSS has wires/vias outside die area`.
They seem to happen during grt stage, after cts is finished.

We do seem to have the power grid outside the die area:

[image]
 

### Expected Behavior

Not print warnings.

### Environment

```shell
They are happening on master, in jenkins also.
```

### To Reproduce

download the artifacts from jenkins nightly.

gudeh, 2024-06-26:
If someone would like to point out what can be done to fix, I can try it myself.

maliberty, 2024-06-26:
It looks like the pdn setup for the outside ring has a core offset that exceeds the core spacing in the design config

gudeh, 2024-06-26:
I tried changing the following parameters but I see the same PDN after floorplan:
[image]

gudeh, 2024-06-26:
This design's config.mk file is the following, I am not sure if this parameters would impact the PDN:
```
export DESIGN_NICKNAME = aes
export DESIGN_NAME = aes_cipher_top
export PLATFORM    = ihp-sg13g2

export VERILOG_FILES = $(sort $(wildcard ./designs/src/$(DESIGN_NICKNAME)/*.v))
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/constraint.sdc

export CORE_UTILIZATION = 20 
export CORE_ASPECT_RATIO = 1
export CORE_MARGIN = 2

export PLACE_DENSITY = 0.65
export TNS_END_PERCENT = 100

export USE_FILL = 1

```

gudeh, 2024-06-27:
This is what we get with master now:
[image]

closed 2024-06-27 by gudeh