The-OpenROAD-Project/OpenROAD #5291: Integrate repair_design and global_placement

maliberty, 2024-06-26:
### Description

Running repair_design after global_placement can lead to a placement density hotspots that deteriorate results.

### Suggested Solution

Early in global placement the positions are subject to too much change to commit to the actual buffering/sizing of repair_design.  As the placement overflow gets low enough and the placement is stabilizing it makes sense to switch to actually modifying the design so that the density can be seen.

This will require significant tuning to determine the right point at which to switch over and also to ensure that multiple calls to repair operate well.

@gadfort @QuantamHD @rovinski for any opinions.  I want to align on the goal before we get too far into implementation.

maliberty, 2024-06-27:
@rovinski in at least some designs I've seen that the repairs are concentrated in local areas.  I also considered doing some global adjustment for the amount of area predicted during virtual resizer but I'm concerned that it won't resolve local issues.

gudeh, 2024-06-27:
Hi everyone. This is what happens on GPL timing-driven placement: during GPL, we call timing-driven based on a list of 6 fixed values of overflow, defined in src/gpl/src/replace.tcl: `set overflow_list [list 79 64 49 29 21 15]`.

When a timing-driven iteration happens, the resizer does all its work and afterwards removes all the work done (we call it a virtual resizing), such as resizing, buffer insertion, and pin swapping (possibly more modifications, I am not sure yet). What is kept in GPL are only the slack weights calculated by the resizer's work and placement changes based on the slack weights only (due to timing-driven adjustments).

I tried a first experiment by keeping the work done by the resizer only on the last overflow value (15), meaning the last timing-driven iteration is not a virtual one, and we get a nice density correlation between steps 3_3_place_gp and 3_4_place_resized.
[image] [image]

We currently have a mismatch between the same steps in master:
[image] [image]

gudeh, 2024-11-25:
This issue was solved with PR: https://github.com/The-OpenROAD-Project/OpenROAD/pull/6165

Other PR changes also went together. To implement this functionality without runtime slow downs we changed a vector access from pointers to indexes using a special handle (since there is a vector that points to two different storage vectors), this  handle was implemented  in PR: https://github.com/The-OpenROAD-Project/OpenROAD/pull/5923

We opted to maintain the 6 fixed timing-driven iterations based on overflow thresholds, although now we do 3 TD virtual iterations (undo all work by rsz at the iteration end), and afterward we do 3 non-virtual TD iterations (keep rsz changes). [This PR](https://github.com/The-OpenROAD-Project/OpenROAD/pull/6067) only introduces a TCL command for the user to modify the number of virtual/non-virtual timing-driven iterations.

closed 2024-11-25 by maliberty

gudeh, 2024-11-25:
Here there is some images showing placement density before the change (top) and after the change (bottom): https://drive.google.com/drive/folders/1vmZFFemVsjTfid3QStEZ5TrWDtVBwnlY?usp=drive_link

Here I did some experiments trying different non-virtual iterations (1, 2 and 3): https://drive.google.com/drive/folders/1U4u6opRNsjoS8MwOxLdjzQMA5LNFUdb2?usp=drive_link