The-OpenROAD-Project/OpenROAD #5294: grt error with high density aes

maliberty, 2024-06-27:
### Describe the bug

If designs/nangate45/aes/config.mk is modified to:

```
export DESIGN_NICKNAME = aes
export DESIGN_NAME = aes_cipher_top
export PLATFORM    = nangate45

export VERILOG_FILES = $(sort $(wildcard ./designs/src/$(DESIGN_NICKNAME)/*.v))
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/constraint.sdc

#export FLOORPLAN_DEF = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/aes_ng45_fp.def

export GPL_TIMING_DRIVEN = 0
export GPL_ROUTABILITY_DRIVEN = 0
export CORE_UTILIZATION = 75
export PLACE_DENSITY = 0.95
export TNS_END_PERCENT        = 100
#export EQUIVALENCE_CHECK     ?=   1
export REMOVE_CELLS_FOR_EQY   = TAPCELL*
```

then grt generates
```
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0170] Net _06047_: Invalid index for position (174300, 312900). Net degree: 176.
[WARNING GRT-0153] Net _06047_ has errors during updateRouteType2.
```

### Expected Behavior

No warnings

### Environment

```shell
OR: 343d3c092e780805be3af44a9fa248679bd4ec77
ORFS: 204610eaf40c1d4389d776a3bdc9dfa21554727c
```

### To Reproduce

run as above

eder-matheus, 2024-06-27:
The Steiner tree has a loop right in the position of the warning:
[image]

The loop is propagated to the rectilinear tree:
[image]

After the congestion iterations, the rectilinear tree is fixed, with no loops on it:
[image]
I believe the loop is fixed by the update @luis201420 made some months ago.

The 3D tree looks fine too:
[image]

eder-matheus, 2024-06-27:
@maliberty I wonder if these warnings should be changed to debug messages. There is nothing the user can do about it, and they are more useful when debugging a problem caused by the loops. And since the tool can recover and fix the loops, the warnings pollute the log.

closed 2024-06-27 by maliberty

maliberty, 2024-06-27:
I think the move to debug is ok though it is a strange crossing.  Its not quite the usual "knock-knee" problem.

maliberty, 2024-06-28:
Why this net is so ugly is also probably worth a peak