The-OpenROAD-Project/OpenROAD #5305: Huge runtime for AES sky130hd

gadfort, 2024-06-28:
### Describe the bug

In the CI runner, the AES sky130hd takes 3+ hours to complete, while other platforms are < 30 mins (except for ihp130, which is 1hour+).
This appears to be related to the large number of shorts reported after iteration 0: 22k+ (which take 50 iterations to resolve, a little over 2 hours), the same can be said for ihp130.

### Expected Behavior

Routing runtimes based on the selected density settings that are comparable with other platforms with a lower number of initial shorts.

### Environment

```shell
Jenkins CI env.
```

### To Reproduce

Observed via jenkins interface, should be reproducible.

### Relevant log output

```shell
https://jenkins.openroad.tools/blue/rest/organizations/jenkins/pipelines/OpenROAD-flow-scripts-Public/pipelines/public_tests_all/branches/master/runs/1285/nodes/123/steps/1227/log/?start=0
```

maliberty, 2024-06-29:
This looks to be drt slowing down due to a tighter placement
[image]

vs

[image]

gadfort, 2024-06-29:
@maliberty I'm concerned with the number of shorts that DRT is reporting (ihp130 has the same issue). This problem has been around for a while is the reason for sky130, gf180, and ihp130 having larger routing times than others. I think it's worth exploring if this is a routing guide issue or a DRT issue. Placement wise this doesn't look like it would require 22k shorts.

maliberty, 2024-06-29:
Ok let's look at both

gudeh, 2024-07-01:
> This looks to be drt slowing down due to a tighter placement [...]

Hi @maliberty, this prints are from which versions exactly? Is it rudy vs grt on routability?

maliberty, 2024-07-01:
The tighter one is rudy.  I didn't save the version but it was just a few days ago I don't think it will have changed much.

gudeh, 2024-07-02:
I tried inflating a lot more during gpl routabiltiy and this is the placement:
[image]

Although we still get a lot of DRCs in the beginning of DRT (left is master, right is more inflation):
[image]

Notice a really high elapsed time. I am waiting for it to finish so I can investigate the DRCs.

maliberty, 2024-07-02:
Would you include the placement image at drt as it is possible more instances where inserted post-gpl that are causing the problem and it can't be fixed in gpl.

gudeh, 2024-07-02:
from make gui_place: 
[image]

from  make gui_5_1_grt:
[image]

And routing congestion from make gui_5_1_grt:
[image]

I will send from DRT after it finishes.

maliberty, 2024-07-02:
drt won't change the placement so that's fine.  It doesn't appear to be a post-gpl increase issue.

gudeh, 2024-07-03:
For some reason DRT is running sequentially on my gcloud machine. It has been running for a whole day, and each 10% of an iteration is taking from 10min to 1 hour.

Either way, here are the DRCs after inflating more on gpl.
[image]

Here are some of the DRCs. 
[image]

I can provide more information if needed.

gudeh, 2024-07-04:
Using grt on routability:
[image]

Using rudy with more inflation than master:
[image]

Current master with rudy and less inflation:
[image]

eder-matheus, 2024-07-22:
@gadfort I've changed the global routing resources configuration for sky130hd/aes, reducing more resources than before, and DRT completes the design in 9 minutes. Similar to what happened with sky130hs/riscv32i in PR https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2142, it also had multiple shorts that were created due to GRT assigning too many nets to the same region.

We're tracking this issue in https://github.com/The-OpenROAD-Project/OpenROAD/issues/5391, so can this issue be closed?

osamahammad21, 2024-08-04:
Update: I removed the drt tag since this doesn't seem to be drt issue.

gudeh, 2025-01-02:
> [@eder-matheus](https://github.com/eder-matheus) this is also tracking issues from GPL, so we cannot close this.

Hi @gadfort, would mind explaining exactly what you see as an issue concerning gpl?

gudeh, 2025-01-02:
Something that seems unusual for sky130hd/aes is that it has some placement density hotspots after placement. 
[image]

After my recent GPL modification, where we keep the changes made by RSZ repair_design during timing-driven iterations (PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/6165)), none of the other designs showed density hotspots—sky130hd/aes is the only one still exhibiting them!

The hotspots are located along the borders of the routing congestion regions, suggesting that routability mode may be moving the cells excessively during the inflation procedure. This behavior does not occur in other designs. On the current master branch, the first routability iteration is applying a +92% increase in area, with one iteration even reaching a 292% increase in area!

Does any of this relate to your concern regarding GPL on sky130hd/aes?