The-OpenROAD-Project/OpenROAD #5306: cts not honoring fanout limits

gadfort, 2024-06-29:
### Describe the bug

CTS does not appear to be honoring fanout limits in ihp130.
```
Pin                                   Limit Fanout  Slack
---------------------------------------------------------
clkbuf_leaf_11_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_12_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_16_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_29_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_41_clk/X                      8     14     -6 (VIOLATED)
```

### Expected Behavior

No DRV violations from CTS

### Environment

```shell
Jenkins CI runner
```

### To Reproduce

aes with ihp130 from ORFS (in jenkins CI so should be reprodusable)

### Relevant log output

```shell
Pin                                   Limit Fanout  Slack
---------------------------------------------------------
clkbuf_leaf_11_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_12_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_16_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_29_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_41_clk/X                      8     14     -6 (VIOLATED)
clkbuf_leaf_9_clk/X                       8     14     -6 (VIOLATED)
clkbuf_leaf_13_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_17_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_20_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_28_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_40_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_48_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_53_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_55_clk/X                      8     13     -5 (VIOLATED)
clkbuf_leaf_5_clk/X                       8     13     -5 (VIOLATED)
clkbuf_leaf_19_clk/X                      8     12     -4 (VIOLATED)
clkbuf_leaf_22_clk/X                      8     12     -4 (VIOLATED)
clkbuf_leaf_23_clk/X                      8     12     -4 (VIOLATED)
clkbuf_leaf_25_clk/X                      8     12     -4 (VIOLATED)
clkbuf_leaf_35_clk/X                      8     12     -4 (VIOLATED)
clkbuf_leaf_46_clk/X                      8     12     -4 (VIOLATED)
clkbuf_leaf_54_clk/X                      8     12     -4 (VIOLATED)
clkbuf_leaf_0_clk/X                       8     11     -3 (VIOLATED)
clkbuf_leaf_15_clk/X                      8     11     -3 (VIOLATED)
clkbuf_leaf_1_clk/X                       8     11     -3 (VIOLATED)
clkbuf_leaf_21_clk/X                      8     11     -3 (VIOLATED)
clkbuf_leaf_34_clk/X                      8     11     -3 (VIOLATED)
clkbuf_leaf_52_clk/X                      8     11     -3 (VIOLATED)
clkbuf_leaf_57_clk/X                      8     11     -3 (VIOLATED)
clkbuf_3_1_0_clk/X                        8     10     -2 (VIOLATED)
clkbuf_3_2_0_clk/X                        8     10     -2 (VIOLATED)
clkbuf_leaf_10_clk/X                      8     10     -2 (VIOLATED)
clkbuf_leaf_2_clk/X                       8     10     -2 (VIOLATED)
clkbuf_leaf_32_clk/X                      8     10     -2 (VIOLATED)
clkbuf_leaf_38_clk/X                      8     10     -2 (VIOLATED)
clkbuf_leaf_39_clk/X                      8     10     -2 (VIOLATED)
clkbuf_leaf_45_clk/X                      8     10     -2 (VIOLATED)
clkbuf_leaf_51_clk/X                      8     10     -2 (VIOLATED)
clkbuf_leaf_7_clk/X                       8     10     -2 (VIOLATED)
clkbuf_leaf_8_clk/X                       8     10     -2 (VIOLATED)
clkbuf_3_0_0_clk/X                        8      9        (VIOLATED)
clkbuf_3_6_0_clk/X                        8      9        (VIOLATED)
clkbuf_leaf_30_clk/X                      8      9        (VIOLATED)
clkbuf_leaf_31_clk/X                      8      9        (VIOLATED)
clkbuf_leaf_33_clk/X                      8      9        (VIOLATED)
clkbuf_leaf_3_clk/X                       8      9        (VIOLATED)
clkbuf_leaf_42_clk/X                      8      9        (VIOLATED)
```

maliberty, 2024-06-29:
I don't see such errors in the latest CI (https://jenkins.openroad.tools/blue/rest/organizations/jenkins/pipelines/OpenROAD-flow-scripts-Nightly-Public/runs/4174/nodes/1246/log/?start=0)

gadfort, 2024-06-29:
@maliberty it's in the timing report (for CTS) in the uploaded artifacts (I'm not sure why the CI is not showing the timing in the log).

maliberty, 2024-06-29:
@precisionmoon I don't see anything in CTS that checks max fanout.  I tried setting `export CTS_CLUSTER_SIZE = 8` but that only fixes the leaf clusters and violations remain higher in the tree.

gadfort, 2024-10-18:
Fixed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/5516

closed 2024-10-18 by gadfort