The-OpenROAD-Project/OpenROAD #5325: Repair Antennas non-deterministic/faulty results

osamahammad21, 2024-07-02:
### Describe the bug

In https://github.com/The-OpenROAD-Project/OpenROAD/pull/5281 The CI is giving different result for sky130hd_ibex than my local machine run. The differences start at repair_antennas part, while the groute guides are identical. What's more weird is that at the end of repair_antennas, the number of antenna violations reported is 0 while after check antennas which is called right after it, the antenna violations are 1 (This is on the CI's result). 

### Expected Behavior

Same number of antenna violations reported by repair_antennas and check_antennas
Same result on my machine and the CI's

### Environment

```shell
kernel: Linux 6.5.0-41-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-14378-gc1c65bb8c
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 98 lines cut ...]
-- TCL header: /usr/include/tcl/tcl.h
-Number of processor cores: 16
- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.0Dd7peA3Ac
```

### To Reproduce

This is from a branch on my fork https://github.com/osamahammad21/OpenROAD/tree/drt-fix-update-design
 This is a non-determinism issue so I am not sure if this happens specifically on my local machine or does the difference appear on other machines as well. What's concerning more tho and I think would be the root of the problem is the difference in the antenna violations number we get from check_antennas and repair_antennas. 

### Screenshots

[image]

Left is my local machine and right is on the CI machine

[image]

gudeh, 2024-07-03:
The previous issue #5233 seems to be due to the same reason.

I noticed this when running the regression test in `./test/regression jpeg_sky130hd`.

I ran with valgrind on a previous OR version and got the following log:

```
################################################################
# Antenna repair
repair_antennas -iterations 5
==5208== Invalid read of size 8
==5208==    at 0x331ED57: odb::tmg_conn::splitBySj(int, odb::tmg_rc_sh const*, int, int, int, int, int) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x331F4D9: odb::tmg_conn::splitTtop() (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x3322024: odb::tmg_conn::findConnections() [clone .part.0] [clone .constprop.0] (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x3324BA5: odb::tmg_conn::analyzeNet(odb::dbNet*) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2DA60D5: grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2DA7314: grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2DA784D: grt::RepairAntennas::checkAntennaViolations(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int, odb::dbMTerm*, float) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2D938D6: grt::GlobalRouter::repairAntennas(odb::dbMTerm*, int, float) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x13B52E3: _wrap_repair_antennas (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x5B93D31: TclNRRunCallbacks (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==5208==    by 0x5B95027: ??? (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==5208==    by 0x5B94AA6: Tcl_EvalEx (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==5208==  Address 0x6e2c5378 is 54,424 bytes inside a block of size 106,496 free'd
==5208==    at 0x5B31B6F: operator delete(void*, unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==5208==    by 0x3325244: void std::vector<odb::tmg_rc, std::allocator<odb::tmg_rc> >::_M_realloc_insert<odb::tmg_rc const&>(__gnu_cxx::__normal_iterator<odb::tmg_rc*, std::vector<odb::tmg_rc, std::allocator<odb::tmg_rc> > >, odb::tmg_rc const&) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x331EBCE: odb::tmg_conn::addRc(int, odb::tmg_rc_sh const&, int, int, int, int, int, int) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x331EF2C: odb::tmg_conn::splitBySj(int, odb::tmg_rc_sh const*, int, int, int, int, int) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x331F4D9: odb::tmg_conn::splitTtop() (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x3322024: odb::tmg_conn::findConnections() [clone .part.0] [clone .constprop.0] (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x3324BA5: odb::tmg_conn::analyzeNet(odb::dbNet*) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2DA60D5: grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2DA7314: grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2DA784D: grt::RepairAntennas::checkAntennaViolations(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int, odb::dbMTerm*, float) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x2D938D6: grt::GlobalRouter::repairAntennas(odb::dbMTerm*, int, float) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==    by 0x13B52E3: _wrap_repair_antennas (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
==5208==  Block was alloc'd at
==5208==    at 0x5B2F013: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==5208==    by 0x331E20E: odb::tmg_conn::tmg_conn(utl::Logger*) (in /home/gudeh/Desktop/gplWithRudy/tools/install/OpenROAD/bin/openroad)
[... cut ...]
[WARNING ANT-0009] Net rle.rz4.amp\[2\] requires more than 10 diodes per gate to repair violations.
[WARNING ANT-0009] Net rle.rz4.amp\[6\] requires more than 10 diodes per gate to repair violations.
[WARNING ANT-0009] Net rle.rz4.size\[2\] requires more than 10 diodes per gate to repair violations.
[INFO GRT-0012] Found 227 antenna violations.
[INFO GRT-0015] Inserted 758 diodes.
[INFO GRT-0054] Using detailed placer to place 69 diodes.
[INFO GRT-0012] Found 44 antenna violations.
[INFO GRT-0015] Inserted 399 diodes.
[INFO GRT-0054] Using detailed placer to place 89 diodes.
```

gudeh, 2024-07-03:
Indeed. This valngrind log is from an older version. I actually got this log from my chat with you @maliberty. But it is interesting that values change in the same place as before, the number of diodes reported by GRT.

eder-matheus, 2024-07-04:
@osamahammad21 I've fixed the mismatch of violations reported here: https://github.com/The-OpenROAD-Project/OpenROAD/pull/5346.

When running in gcp, I see the same results as you see. In every run I get the same output, so at least we have consistency in a same OS and machine.

Could you update the branch of your PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/5281? I can't access the results of the older CI runs, probably because we had some updates in the CI.

eder-matheus, 2024-07-04:
Reopening it to keep investigating the non-determinist across platforms.

osamahammad21, 2024-07-08:
@eder-matheus I updated the branch. It's passing now but I haven't checked if the results are different or the same as the updated metrics limits.

eder-matheus, 2024-07-08:
@osamahammad21 The divergence is not happening anymore. I've run your branch in GCP and compared it with the artifacts from the CI, and the results are now equal.

closed 2024-07-08 by eder-matheus