The-OpenROAD-Project/OpenROAD #5329: Mapping mouse wheel to zoom in/out

titan73, 2024-07-03:
### Description

Currently navigation is done through the arrow keys and the mouse wheel. The latter is basically the same as the up & down keys.
This is redundant and not very useful. 

### Suggested Solution

That would be much practical to use the mouse wheel for zooming in & out like commercial tool (ex: Innovus).

AcKoucher, 2024-07-03:
@titan73 You can zoom in/out with CTRL + mouse wheel. You can also use Z for zoom in and SHIFT+Z for zoom out.

titan73, 2024-07-03:
Ok thanks. It's better if I use the right control close to the arrow keys.
I still think zooming should be the default without using CTRL though. Or adding an option to choose.

stefanottili, 2024-07-03:
I'm also used to "zoom in/out with mouse wheel" which is also the default in klayout.

For klayout have to change key mappings, because I'm used to using z/Z,f for "zoom in/out, fit".
I also change the percentage of scroll for the cursor movements to 40% and fiddle with the mouse sensitivity.

If you look at layoutEditor or glade, they have yet another set of defaults.
Unfortunately there doesn't seem to be a consensus on default key and mouse settings.

The only solution really is to be able to be able configure this, preferably in Settings.

PS: ... and since we're talking preferences ..

People can't even agree on the metal stack colors ;-)

If you started with 3 layer metal like me, then the metal stack will be primary colors, because that's what cell3 and the displays offered in the middle last decade of the last millennium.

M1/2/3 will be red/green/blue and later M4/5/6 yellow/cyan/magenta ... No consensus above that.

And do you color vias differently or like the metal up or down ?
V1/V2 like M1/M2 or use M1 color for CO and M2 color for V1 ?

Once again, the only solution will be to be able to change that in Settings.

titan73, 2024-07-03:
So people use mouse wheel to go up & down and arrow keys to go left & right?
Since up & down keys are close to the left & right ones, that would be better to do something else with the mouse wheel.

maliberty, 2024-07-03:
You can double click the color chip by the layer to configure it.  We don't have an external config file but that would be a reasonable enhancement.

If the community wants to contribute here that would be great as these only require Qt experience and there are many core issues that need experienced developers to tackle.  The GUI is serviceable but could be more elegant.

titan73, 2024-07-04:
I don't understand where I should double click. What is "the color chip by the layer"?

maliberty, 2024-07-04:
[image]

titan73, 2024-07-05:
Ah ok. Sorry. I'm just realizing that this was an answer to someone else. Nothing to do with zooming.

titan73, 2024-07-15:
> You can double click the color chip by the layer to configure it. We don't have an external config file but that would be a reasonable enhancement.
>
> If the community wants to contribute here that [...]

I don't any Qt experience but I can look at it. I modified the code to invert the behaviour of the mouse wheel. By default, zoom in/out and with CTRL scroll up/dows. So what more is needed. A configuration option?

maliberty, 2024-07-15:
Yes a user option.  We could create a preferences form that would have this single option as a start.  It should be reachable from the menus.

closed 2024-07-16 by AcKoucher