The-OpenROAD-Project/OpenROAD #5331: Nothing created with add_pdn_stripe

titan73, 2024-07-03:
### Describe the bug

I just ran the command:
    add_pdn_stripe -layer metal4B -width 5 -pitch 100 -nets {VDDD1V2 VDDD GNDD} -offset 10

The command is executed without any errors but no stripes are created.*

### Expected Behavior

Stripes should be created.

### Environment

```shell
OpenROAD 74da93a32ce0cb78d8d93ca473121b8cdd52ff8d from openroad-2024-06-11 package.
```

### To Reproduce

initialize_floorplan -die_area {0 0 500 500} -core_area {0 0 500 500} -site CORECUBRICK
set db [ord::get_db_block]
set gnd_net [odb::dbNet_create $db VDDD]
set gnd_net [odb::dbNet_create $db GNDD]
add_pdn_stripe -layer metal4B -width 5 -pitch 100 -nets {VDDD GNDD} -offset 10

eder-matheus, 2024-07-03:
Please add the files of your test case, if possible. We can't reproduce your issue just with the script you used.

titan73, 2024-07-03:
Ah indeed. I should have taken the full example at the bottom of https://openroad.readthedocs.io/en/latest/main/src/pdn/README.html. Thanks.

closed 2024-07-03 by maliberty