The-OpenROAD-Project/OpenROAD #5340: script output display is very slow when the gui is on

titan73, 2024-07-04:
### Describe the bug

I've noticed a slowness in gui which is very annoying. It seems this comes from the console input & output.

Using the following procedure:

    proc measure_time {} {
        set TIME_start [clock clicks -milliseconds]
        for {set i 0} {$i < 100} {incr i} {
            puts "Iteration $i"
        }
        set TIME_taken [expr [clock clicks -milliseconds] - $TIME_start]
        puts "Duration $TIME_taken ms"
    }

Without gui the script takes less that 1 ms.

With gui (-gui option or gui::show), it takes 40 ms.
    - source the script takes less that 1 ms whenever it is called.

This is another specific condition in gui that show the slowness with the procedure taking 8 ms.

Basically after link_design is called, all subsequent commands before you get the prompt again are horribly slow.

So basically a script test.tcl ending by run by openroad -gui test.tcl :

    measure_time => 40 ms
    link_design test
    measure_time => 8 s

Once the prompt is available calling again measure_time will takes 40 ms

Now commenting these 3 lines and start again, executing these 3 commands one by one => 40 ms

Same with 3 commands called in a row will give 40 ms followed by 8 s.

### Expected Behavior

Should be few.

### Environment

```shell
OpenROAD 74da93a32ce0cb78d8d93ca473121b8cdd52ff8d from openroad-2024-06-11 package
```

### To Reproduce

openroad -gui test.tcl

test.tcl:

    proc measure_time {} {
        set TIME_start [clock clicks -milliseconds]
        for {set i 0} {$i < 100} {incr i} {
            puts "Iteration $i"
        }
        set TIME_taken [expr [clock clicks -milliseconds] - $TIME_start]
        puts "Duration $TIME_taken ms"
    }
    ....
    measure_time
    link_design test
    measure_time

I'm trying to make a test case. It does not seem easy as using a simpler design does not show the issue.

titan73, 2024-07-10:
I'm trying to do a test case with GF180. In openroad there is the file gf180mcu_5LM_1TM_9K_9t_sc.lef for the standard cells. Where can I find the .lib files associated?

maliberty, 2024-07-10:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/platforms/gf180/lib

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/gf180/lef/gf180mcu_5LM_1TM_9K_9t_tech.lef

titan73, 2024-07-11:
Thanks. Here is a test case:
    openroad test.tcl
without and with -gui

In this test, in gui mode, I get only one message after the command link (TIME: duration ....) then nothing.

[slow.zip](https://github.com/user-attachments/files/16179674/slow.zip)

titan73, 2024-08-08:
I tested today's git and the problem still happens. After link_design, traces are not displayed and openroad gets killed.

I compiled:

Commit 628069c56b99068f76d35e06879a786b2e7e3047 (HEAD -> master, upstream/master)
Merge: db7d56e4d 1114f30a9
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Wed Aug 7 17:26:21 2024 -0700
    Merge pull request #5528 from The-OpenROAD-Project/patch-cpp20
    Update Dockerfile.cpp20

Which is after the PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/5514 was merged.

Tested with the test case provided.

titan73, 2024-08-09:
Retested with today's git and same. The issue is not fixed. Can you reopen this issue?

AcKoucher, 2024-08-16:
Some debugging suggests that there's some sort of deadlock happening with the GuiSink. The issue here is not related to slowness anymore. I opened #5600 for it.

PR #5514 (2024-08-06): gui: use a buffer to store report text to avoid slowing down OR when logging in Scripting
Resolve #5340
closed 2024-08-16 by AcKoucher