The-OpenROAD-Project/OpenROAD #5343: Behaviour difference between console and gui modes when reading .def twice

titan73, 2024-07-04:
### Describe the bug

When reading .def (workaround needed to force creation of power ports until upf support is improved), the behaviour is not the same between console and gui.

Basically I have this in my tcl.

    puts "DEF 1:"
    read_def -floorplan_initialize def.def
    puts "DEF 2:"
    read_def -floorplan_initialize def.def

In gui mode, it works and I get:

    DEF 1:
    [INFO ODB-0127] Reading DEF file: def.def
    [INFO ODB-0128] Design: test
    [INFO ODB-0253]     Updated 1 components.
    [INFO ODB-0134] Finished DEF file: def.def
    DEF 2:
    [INFO ODB-0127] Reading DEF file: def.def
    [INFO ODB-0128] Design: test
    [INFO ODB-0253]     Updated 1 components.
    [INFO ODB-0134] Finished DEF file: def.def

In console mode, if failed the 2nd time with:

    DEF 1:
    [INFO ODB-0127] Reading DEF file: def.def
    [INFO ODB-0128] Design: test
    [INFO ODB-0253]     Updated 1 components.
    [INFO ODB-0134] Finished DEF file: def.def
    DEF 2:
    [INFO ODB-0127] Reading DEF file: def.def
    [INFO ODB-0128] Design: test
    [ERROR ODB-0359] Attempt to change the origin of FIRM instance NVM_512bE2LTa_H9A_i
    Error: def.tcl, 18 ODB-0359

The user interface (gui or console) should not alter the behaviour of the commands.

### Expected Behavior

2nd read_ref command should success.

### Environment

```shell
OpenROAD 74da93a32ce0cb78d8d93ca473121b8cdd52ff8d from openroad-2024-06-11 package.
```

### To Reproduce

I'll make a testcase when I find a usable set of lefs (technolofy, sites and standard cells) from OR and ORFS gits (sky130 or gf180).

maliberty, 2024-07-05:
> 2nd read_ref command should success.

I think it should fail.  What is the goal of reading it twice?  I think the GUI should error as well.

maliberty, 2024-07-05:
They both pass now.  There was an oddity to the check for placing an instance at its current location (which happens on the 2nd read).

closed 2024-07-05 by maliberty

titan73, 2024-07-05:
I reread the .def a s second time to get the power pins created. I's a workaround until the upf support is improved.
I mentioned it here https://github.com/The-OpenROAD-Project/OpenROAD/discussions/5289#discussioncomment-9949407

PR #5347 (2024-07-05): odb: correct the check in dbInst::setOrigin for no-op
Fixes #5343
commit 03dd99db (2024-07-05): odb: correct the check in dbInst::setOrigin for no-op

Fixes #5343