The-OpenROAD-Project/OpenROAD #5355: read_power_activities warnings

oharboe, 2024-07-08:
### Describe the bug

Based on https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2113

1. extract [mock-array-power.tar.gz](https://github.com/user-attachments/files/16128226/mock-array-power.tar.gz)
2. `./run-me-mock-array-asap7-base.sh`

Warnings:

```
OpenROAD v2.0-14390-gf2a21152b 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
read_spef ./results/asap7/mock-array/base/6_final.spef
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.27e-04   0.00e+00   1.03e-08   3.27e-04   8.2%
Combinational          1.38e-13   8.49e-14   2.36e-07   2.36e-07   0.0%
Clock                  1.77e-03   1.89e-03   1.49e-08   3.66e-03  91.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.10e-03   1.89e-03   2.61e-07   3.99e-03 100.0%
                          52.6%      47.4%       0.0%
[WARNING STA-1451] problem parsing bus io_outs_right_5.
[WARNING STA-1451] problem parsing bus io_outs_right_4.
[WARNING STA-1451] problem parsing bus io_outs_right_7.
[WARNING STA-1451] problem parsing bus io_outs_right_6.
[WARNING STA-1451] problem parsing bus io_outs_right_1.
[WARNING STA-1451] problem parsing bus io_outs_right_0.
[WARNING STA-1451] problem parsing bus io_outs_right_3.
[WARNING STA-1451] problem parsing bus io_outs_right_2.
[WARNING STA-1451] problem parsing bus io_ins_left_7.
[deleted]
```

### Expected Behavior

No warnings

### Environment

```shell
OpenROAD v2.0-14390-gf2a21152b
```

### To Reproduce

See above

oharboe, 2024-07-08:
Looks fine in .vcd file and in gtkwave...

```
[deleted]
   $var wire 64 $Z io_ins_left_7 $end
[deleted]
b0000000000000000000000000000000000000000000000000000000000000000 $Z
[deleted]
b0000000000000000000000000000000000000000000000000000000000000111 $Z
[deleted]
```

[image]

maliberty, 2024-07-08:
Please try with Verilator at gate-level

oharboe, 2024-07-09:
Looking into this build error now... https://github.com/verilator/verilator/issues/5244

oharboe, 2024-07-10:
Post synthesis simulation .vcd file (from 6_final.vcd)

unzip [read-power-activites-post-synthesis.zip](https://github.com/user-attachments/files/16157886/read-power-activites-post-synthesis.zip)

No activities read:

```
$ ./run-me-mock-array-asap7-base.sh 
OpenROAD v2.0-14454-g8d52b96f9 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
read_spef ./results/asap7/mock-array/base/6_final.spef
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.27e-04   0.00e+00   1.03e-08   3.27e-04   8.2%
Combinational          1.38e-13   8.49e-14   2.36e-07   2.36e-07   0.0%
Clock                  1.77e-03   1.89e-03   1.49e-08   3.66e-03  91.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.10e-03   1.89e-03   2.61e-07   3.99e-03 100.0%
                          52.6%      47.4%       0.0%
Annotated 0 pin activities.
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.27e-04   0.00e+00   1.03e-08   3.27e-04   8.2%
Combinational          1.38e-13   8.49e-14   2.36e-07   2.36e-07   0.0%
Clock                  1.77e-03   1.89e-03   1.49e-08   3.66e-03  91.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.10e-03   1.89e-03   2.61e-07   3.99e-03 100.0%
                          52.6%      47.4%       0.0%
openroad> 
```

end of gui.tcl was updated in the reproduction case to run read_power_activities.

```
$ tail scripts/gui.tcl 
    }
  }
  
  # Cleanup temporary variables
  unset sdc_file s design_stage
}

report_power
read_power_activities -vcd MockArrayTestbench.vcd -scope MockArrayTestbench/postSynthesis
report_power
```

According to gtkwave, there are some pin activities...

[image]

maliberty, 2024-07-11:
You are using `-scope MockArrayTestbench/postSynthesis` but your vcd has
```
 $scope module TOP $end
```
so you should use -scope TOP.   With that `Annotated 4162 pin activities.` and there is a small change in combinational power.

oharboe, 2024-07-12:
When I do:

```
$ ./run-me-mock-array-asap7-base.sh 
blah
OpenROAD v2.0-14506-ge680971bb 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
read_spef ./results/asap7/mock-array/base/6_final.spef
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.27e-04   0.00e+00   1.03e-08   3.27e-04   8.2%
Combinational          1.38e-13   8.49e-14   2.36e-07   2.36e-07   0.0%
Clock                  1.77e-03   1.89e-03   1.49e-08   3.66e-03  91.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.10e-03   1.89e-03   2.61e-07   3.99e-03 100.0%
                          52.6%      47.4%       0.0%
Annotated 46598 pin activities.
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.27e-04   0.00e+00   1.03e-08   3.27e-04   8.2%
Combinational          1.50e-14   8.95e-15   2.86e-07   2.86e-07   0.0%
Clock                  1.77e-03   1.89e-03   1.49e-08   3.66e-03  91.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.10e-03   1.89e-03   3.11e-07   3.99e-03 100.0%
                          52.6%      47.4%       0.0%
openroad> 
```

maliberty, 2024-07-12:
close?

oharboe, 2024-07-12:
@maliberty More questions... This doesn't include the 64 Elements, how do I get those into the report_power?

oharboe, 2024-07-12:
The power activities of each element is within this .vcd file, so I "just" need to add in those power activities.

How do I combine this into a single report_power?

[read-power-element.tar.gz](https://github.com/user-attachments/files/16189398/read-power-element.tar.gz)

With the modified version above I read in the Element 6_final.odb and I run:

```
read_power_activities -vcd MockArrayTestbench.vcd -scope TOP/MockArrayTestbench/postSynthesis
```

I then get 527 pin activities and a significant *decrease* in combinational power:

```
OpenROAD v2.0-14513-g943ac03e1 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0347] current_design for other than top cell not supported.
[WARNING STA-0366] port 'io_ins_down_0[0]' not found.
[WARNING STA-0366] port 'io_ins_down_0[10]' not found.
[WARNING STA-0366] port 'io_ins_down_0[11]' not found.
[WARNING STA-0366] port 'io_ins_down_0[12]' not found.
[deleted]
[WARNING STA-1650] message limit reached, this message will no longer print
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.25e-03   8.68e-05   9.25e-08   3.34e-03  61.5%
Combinational          9.89e-05   3.33e-05   5.68e-08   1.32e-04   2.4%
Clock                  1.05e-03   9.07e-04   9.90e-09   1.96e-03  36.1%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  4.40e-03   1.03e-03   1.59e-07   5.43e-03 100.0%
                          81.1%      18.9%       0.0%
Annotated 527 pin activities.
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.14e-03   9.50e-06   1.13e-07   3.15e-03  61.4%
Combinational          1.45e-05   4.61e-06   7.58e-08   1.91e-05   0.4%
Clock                  1.05e-03   9.07e-04   9.90e-09   1.96e-03  38.3%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  4.20e-03   9.21e-04   1.99e-07   5.12e-03 100.0%
                          82.0%      18.0%       0.0%
```

oharboe, 2024-07-16:
next stumbling block for mock-array and Element read_power_activities https://github.com/The-OpenROAD-Project/OpenROAD/issues/5383

closed 2024-07-16 by oharboe