The-OpenROAD-Project/OpenROAD #5364: OpenROAD crashes when launching global_route command

titan73, 2024-07-10:
### Describe the bug

After creating the floorplan I do;
    global_placement
    detailed_placement
    clock_tree_synthesis -buf_list {...}

Then launching global_route crashes OpenROAD.

During cts, I have warnings;
    - 1 [WARNING CTS-0104] Clock wire resistance/capacitance values are zero.
    - several [WARNING CTS-0041] Net "xxx" has 1 sinks. Skipping...
    - several [WARNING CTS-0105] Net "xxx" already has clock buffer yyy. Skipping...

### Expected Behavior

No crash.

### Environment

```shell
Ubuntu 22.04
OpenROAD:
    - v2.0-14452-g15e1bfb5e built from git
    - 74da93a32ce0cb78d8d93ca473121b8cdd52ff8d prebuilt from 2024-06-11
```

### To Reproduce

I'll try to make a test case.

### Relevant log output

```shell
openroad> global_route
Signal 11 received
Stack trace:
 0# 0x00005B698AA9FA4B in /home/costa/OpenROAD/build/src/openroad
 1# 0x00007A13F7242520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbTechLayer::getType() in /home/costa/OpenROAD/build/src/openroad
 3# grt::GlobalRouter::calcLayerPitches(int) in /home/costa/OpenROAD/build/src/openroad
 4# grt::GlobalRouter::initRoutingTracks(int) in /home/costa/OpenROAD/build/src/openroad
 5# grt::GlobalRouter::initFastRoute(int, int) in /home/costa/OpenROAD/build/src/openroad
 6# grt::GlobalRouter::globalRoute(bool, bool, bool) in /home/costa/OpenROAD/build/src/openroad
 7# 0x00005B698B1A5D0B in /home/costa/OpenROAD/build/src/openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x00007A13FB890028 in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# ord::tclAppInit(Tcl_Interp*) in /home/costa/OpenROAD/build/src/openroad
13# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# main in /home/costa/OpenROAD/build/src/openroad
15# 0x00007A13F7229D90 in /lib/x86_64-linux-gnu/libc.so.6
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# _start in /home/costa/OpenROAD/build/src/openroad
```

titan73, 2024-07-12:
The test case is not clean but it's enough to reproduce the same crash.

[Uploading crash_grt.zip…]()

titan73, 2024-07-12:
The test case is not clean but it's enough to reproduce the same crash.
[Uploading slow.zip…]()

titan73, 2024-07-12:
File upload at https://filebin.net/ce0ol3qp5x4j42kh as github is still uploading the file for unknown reason.

eder-matheus, 2024-07-24:
@titan73 The fix for the issue was merged, and the cause of the crash is that your input DEF doesn't have routing tracks defined, which is essential for the global router. We added an error message to avoid the crash, and you can try using the make_tracks command: https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/ifp#make-tracks

closed 2024-07-24 by eder-matheus

titan73, 2024-07-26:
@eder-matheus The test case does not read an input DEF but uses initialize_floorplan.
Thanks for the fix. Use make_tracks make global_route works. Thanks!

PR #5442 (2024-07-24): grt: adding error message when no routing layer has a track structure
Adding error message when no routing layer has a track structure.
Fixes #5364