The-OpenROAD-Project/OpenROAD #5365: Crash on "odb::dbInst_getMaster [odb::dbBlock_findInst $db foobar"

titan73, 2024-07-10:
### Describe the bug

While trying to create a test case for https://github.com/The-OpenROAD-Project/OpenROAD/issues/5364. I found another crash.

read_lef gf180mcu_5LM_1TM_9K_9t_tech.lef
read_verilog test.v
link_design test
set db [ord::get_db_block]
odb::dbInst_getMaster [odb::dbBlock_findInst $db foobar]

### Expected Behavior

No crash

### Environment

```shell
Ubuntu 22.04
OpenROAD v2.0-14452-g15e1bfb5e built from git
```

### To Reproduce

openroad test.tcl
[crash.zip](https://github.com/user-attachments/files/16166977/crash.zip)

### Relevant log output

```shell
Signal 11 received
Stack trace:
 0# 0x00005CAE9DA0FA4B in ../OpenROAD/build/src/openroad
 1# 0x0000786888E42520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbInst::getMaster() const in ../OpenROAD/build/src/openroad
 3# 0x00005CAE9DF4DC86 in ../OpenROAD/build/src/openroad
 4# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 5# 0x000078688D4D6028 in /lib/x86_64-linux-gnu/libtcl8.6.so
 6# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
 7# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in ../OpenROAD/build/src/openroad
 9# ord::tclAppInit(Tcl_Interp*) in ../OpenROAD/build/src/openroad
10# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# main in ../OpenROAD/build/src/openroad
12# 0x0000786888E29D90 in /lib/x86_64-linux-gnu/libc.so.6
13# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
14# _start in ../OpenROAD/build/src/openroad
Segmentation fault (core dumped)
```

maliberty, 2024-07-10:
Did `odb::dbBlock_findInst $db foobar` return null?  The odb APIs assume the caller is doing proper operations (its too expensive to check every low level call).

titan73, 2024-07-10:
Yes it returns null. If it's a low level call shouldn't this not supposed to be accessible from the command line then?

maliberty, 2024-07-10:
It is a raw swig of db.h.   It would be more idiomatic to do:
```
>>> set db [ord::get_db_block]
_8065550800000000_p_odb__dbBlock
>>> [odb::dbBlock_findInst $db foobar] getMaster
[ERROR GUI-0070] invalid command name "NULL"
# or
>>> set inst  [odb::dbBlock_findInst $db foobar]
NULL
>>> $inst getMaster
[ERROR GUI-0070] invalid command name "NULL"
```
which avoids this problem.

titan73, 2024-07-12:
test
[Uploading crash_grt.zip…]()

maliberty, 2024-07-12:
We could remove the functionality but it is useful for more expert customization.  It would be too much work to build a wrapper for every API.  If you follow the suggestion above you will be safe.

titan73, 2024-07-12:
I changed the the line from:
    set master [odb::dbMaster_getName [odb::dbInst_getMaster [odb::dbBlock_findInst $db NVM_512bE2LTa_H9A_i]]]
to:
    set master [[[$db findInst NVM_512bE2LTa_H9A_i] getMaster] getName]

I'll stick to this oriented object approach. It's more readable anyway.

That said it seems that odb functions are still necessary. For example function odb::dbNet_create is not a member function of dbNet object but a simple function which takes dbBlock argument and a net name. So a confusion can be easily done with a crash as consequence.

maliberty, 2024-07-12:
Only static functions like dbNet_create need to be called in this style.  In that case there is no object to be null and the arguments can be checked.

titan73, 2024-07-15:
That would be better to use a different namespace for this static functions then unless they are integrated in their respective object (create_net could be a method of dbBlock for instance).
I can live with the current situation as it's an evaluation but from a Company pov, crashes are not good in a production environment (that DC does a bit too often :s)
We can close this ticket.

maliberty, 2024-07-15:
I would like to avoid such crashes as well.  I don't think swig allows us to use a different namespace for statics.  Swig is a hard tool to master so it might have some way but figuring it out is painful.

closed 2024-07-15 by maliberty