The-OpenROAD-Project/OpenROAD #5383: [WARNING STA-0349] instance 'ces_0_0' not found.

oharboe, 2024-07-16:
### Describe the bug

Based on https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2120/commits/113fd8dd5c43002b92661c0ff6fcb882e286f809

1. untar [power.tar.gz](https://github.com/user-attachments/files/16243517/power.tar.gz)
2. run `./run-me-mock-array-asap7-base.sh`

Output:

```
OpenROAD v2.0-14511-gedb4e7e37 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0227] LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells
Reading results/asap7/mock-array_Element/base/6_final.v
Reading ./results/asap7/mock-array/base/6_final.v
[WARNING ORD-2011] LEF master DECAPx10_ASAP7_75t_R has no liberty cell.
[WARNING ORD-2011] LEF master DECAPx6_ASAP7_75t_R has no liberty cell.
[WARNING ORD-2011] LEF master FILLERxp5_ASAP7_75t_R has no liberty cell.
[WARNING ORD-2011] LEF master FILLER_ASAP7_75t_R has no liberty cell.
[WARNING ORD-2011] LEF master DECAPx1_ASAP7_75t_R has no liberty cell.
[WARNING ORD-2011] LEF master DECAPx2_ASAP7_75t_R has no liberty cell.
[WARNING ORD-2011] LEF master DECAPx4_ASAP7_75t_R has no liberty cell.
[WARNING ORD-2011] LEF master TAPCELL_ASAP7_75t_R has no liberty cell.
[WARNING STA-0349] instance 'ces_0_0' not found.
[WARNING STA-0349] instance 'ces_0_1' not found.
[WARNING STA-0349] instance 'ces_0_2' not found.
[deleted]
```

### Expected Behavior

No warnings and able to run read_power_activities from .vcd file.

### Environment

```shell
OpenROAD v2.0-14511-gedb4e7e37
```

### To Reproduce

See above

maliberty, 2024-07-16:
My earlier suggestion was to run this in `sta` not `openroad`.  sta keeps Verilog hierarchy while OR doesn't (currently).  There is no hierarchical ces_0_0 instance in OR to annotate against.  [Technically slightly inaccurate but not in important ways].

oharboe, 2024-07-16:
Does ORFS build sta?

maliberty, 2024-07-16:
Yes, just type `sta`

oharboe, 2024-07-16:
Very good. This issue is closed as such, will open a new issue if I have something more actionable...

tspyrou, 2024-07-16:
@oharboe in results/asap7/mock-array/base/6_final.sdc the set_max_delay exception is pointing to the mock array instance which has now been flattened and does not exist in opendb. I see that you are running in openroad versus standalone sta. If you run in standalone sta this instance will not be flattened and the exception will work.

For standalone sta the script needs to remove openroad specific commands such as read_lef, utl:: and suppress_message, unsuppress_message etc. Once this is done, I see it working perfectly.

remove suppress and unsuppress from liberty.tcl

new power.tcl with openroad specific commands commented out :
#utl::set_metrics_stage "power__{}"

source $::env(SCRIPTS_DIR)/load.tcl

source $::env(SCRIPTS_DIR)/read_liberty.tcl
#read_lef $::env(TECH_LEF)
#read_lef $::env(SC_LEF)

puts "Reading results/asap7/mock-array_Element/base/6_final.v"
read_verilog results/asap7/mock-array_Element/base/6_final.v
puts "Reading $::env(RESULTS_DIR)/6_final.v"
read_verilog $::env(RESULTS_DIR)/6_final.v

link_design MockArray

read_sdc $::env(RESULTS_DIR)/6_final.sdc
# FIXME what about Element
read_spef $::env(RESULTS_DIR)/6_final.spef

report_power
read_power_activities -scope TOP/MockArrayTestbench/postSynthesis -vcd designs/src/mock-array/MockArrayTestbench.vcd
report_power

closed 2024-07-16 by tspyrou