The-OpenROAD-Project/OpenROAD #5384: read_power_activities SEGFAULT

oharboe, 2024-07-16:
### Describe the bug

1. untar [crash.tar.gz](https://github.com/user-attachments/files/16243562/crash.tar.gz)
2. `./run-me-mock-array-asap7-base.sh`

Crash:

```
$ ./run-me-mock-array-asap7-base.sh 
OpenROAD v2.0-14511-gedb4e7e37 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0227] LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells
Signal 11 received
Stack trace:
 0# 0x00005F8D92694553 in openroad
 1# 0x00007B574E042990 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbHashTable<odb::_dbInst>::find(char const*) in openroad
 3# sta::dbSdcNetwork::findInstance(char const*) const in openroad
 4# sta::dbSdcNetwork::findPin(char const*) const in openroad
 5# sta::ReadVcdActivities::setVarActivity(char const*, std::vector<sta::VcdValue, std::allocator<sta::VcdValue> > const&, int) in openroad
[deleted]
```

### Expected Behavior

Error/warning messages

### Environment

```shell
OpenROAD v2.0-14511-gedb4e7e37
```

### To Reproduce

See above

maliberty, 2024-07-16:
While we should protect against the crash, you should note that you haven't loaded any design (eg .def or .v):
```
source $::env(SCRIPTS_DIR)/read_liberty.tcl
read_lef $::env(TECH_LEF)
read_lef $::env(SC_LEF)

read_power_activities -scope TOP/MockArrayTestbench/postSynthesis -vcd designs/src/mock-array/MockArrayTestbench.vcd
```

oharboe, 2024-07-16:
Thanks! Yes I am aware of the problem in the script; I moved up the read_power_activities to verify that the file was found because the subsequent commands take a long time...

maliberty, 2024-07-16:
Had things worked properly we would have errored with no design before even trying to load the file.

oharboe, 2024-07-16:
I see.

Practically, this is just a small nit to convert a segfault to an actionable error...

maliberty, 2024-07-16:
Yes