The-OpenROAD-Project/OpenROAD #5387: Warnings when reading .spef file for Element when trying to do mock-array read_power_activities

oharboe, 2024-07-17:
### Describe the bug

I don't know why the .spef file is read in, but I get warnings when I try to read in the .spef file for Element.

untar [mock-array-sta-warnings.tar.gz](https://github.com/user-attachments/files/16260869/mock-array-sta-warnings.tar.gz)

```
./run-me-mock-array-asap7-base.sh
```

Result:

```
OpenSTA 2.5.0 1c7f022cd0 Copyright (c) 2024, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13178, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13211, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13244, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13277, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13310, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13343, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13376, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14772, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14805, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14838, timing group from output port.
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10511, net _0000_ not found.
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10513, instance io_outs_down\[0\]$_DFF_P_ not found.
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10514, instance _0835_ not found.
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10547, net _0001_ not found.
[delete]
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.26e-04   0.00e+00   1.03e-08   3.27e-04   8.2%
Combinational          1.37e-13   8.30e-14   2.36e-07   2.36e-07   0.0%
Clock                  1.78e-03   1.86e-03   1.50e-08   3.64e-03  91.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.11e-03   1.86e-03   2.61e-07   3.96e-03 100.0%
                          53.2%      46.8%       0.0%
Annotated 46598 pin activities.
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             3.26e-04   0.00e+00   1.03e-08   3.27e-04   8.2%
Combinational          1.51e-14   9.19e-15   2.86e-07   2.86e-07   0.0%
Clock                  1.78e-03   1.86e-03   1.50e-08   3.64e-03  91.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.11e-03   1.86e-03   3.12e-07   3.96e-03 100.0%
                          53.2%      46.8%       0.0%
```

### Expected Behavior

No warnings

### Environment

```shell
OpenSTA 2.5.0 1c7f022cd0
```

### To Reproduce

See above

maliberty, 2024-07-17:
Two problem:

1) You should not load Element.lib as doing so makes it a black box and hides the internals
2) You have to load the parasitics for each element:
```
for {set x 0} {$x < 8} {incr x} {
  for {set y 0} {$y < 8} {incr y} {
    read_spef -path ces_${x}_${y} results/asap7/mock-array_Element/base/6_final.spef
  }
}
```

maliberty, 2024-07-17:
FYI report_parasitic_annotation is helpful to check

oharboe, 2024-07-17:
I will fix this, but can we merge this as a baseline?

This way there is a shared baseline in master for the various efforts and parties involved...

oharboe, 2024-07-18:
> report_parasitic_annotation

I tried the following:

```
$ git diff
diff --git a/flow/designs/asap7/mock-array/power.tcl b/flow/designs/asap7/mock-array/power.tcl
index d80bee9f..505fc42c 100644
--- a/flow/designs/asap7/mock-array/power.tcl
+++ b/flow/designs/asap7/mock-array/power.tcl
@@ -10,9 +10,14 @@ link_design MockArray
 
 read_sdc $::env(RESULTS_DIR)/6_final.sdc
 read_spef $::env(RESULTS_DIR)/6_final.spef
-# FIXME lots of warnings
-#read_spef results/asap7/mock-array_Element/base/6_final.spef
+for {set x 0} {$x < 8} {incr x} {
+  for {set y 0} {$y < 8} {incr y} {
+    puts "Reading ces_${x}_${y}"
+    read_spef -path ces_${x}_${y} results/asap7/mock-array_Element/base/6_final.spef
+  }
+}
 
+report_parasitic_annotation
```

and got:

```
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14838, timing group from output port.
Reading ces_0_0
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10511, net _0000_ not found.
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10513, instance io_outs_down\[0\]$_DFF_P_ not found.
[deleted]
Reading ces_0_1
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10511, net _0000_ not found.
Warning: results/asap7/mock-array_Element/base/6_final.spef line 10513, instance io_outs_down\[0\]$_DFF_P_ not found.
[deleted]
Found 7622 unannotated drivers.
Found 0 partially unannotated drivers.
```

maliberty, 2024-07-18:
I'm guessing you missed "You should not load Element.lib "

oharboe, 2024-07-18:
> I'm guessing you missed "You should not load Element.lib "

What about the PDK .lib files?

oharboe, 2024-07-18:
When I change from LIB_FILES to TC_NLDM_LIB_FILES, the warnings go away as you explained:

```
foreach libFile $::env(TC_NLDM_LIB_FILES) {
read_liberty $libFile
}
```

```
vars-mock-array-asap7-base.sh:export LIB_FILES="${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_AO_RVT_FF_nldm_211120.lib.gz ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_INVBUF_RVT_FF_nldm_220122.lib.gz ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_OA_RVT_FF_nldm_211120.lib.gz ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_SEQ_RVT_FF_nldm_220123.lib  ./results/asap7/mock-array_Element/base/Element.lib"
vars-mock-array-asap7-base.sh:export TC_NLDM_LIB_FILES="${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_AO_RVT_TT_nldm_211120.lib.gz ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_INVBUF_RVT_TT_nldm_220122.lib.gz ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_OA_RVT_TT_nldm_211120.lib.gz ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_SEQ_RVT_TT_nldm_220123.lib ${FLOW_HOME}/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_TT_nldm_211120.lib.gz"
```

maliberty, 2024-07-18:
Probably you can just unset ADDITIONAL_LIBS

maliberty, 2024-07-18:
> So you mean I should read the .lib files from the PDK, but not from the macros?

Yes.  It will get the power data for any std cell from the .lib

oharboe, 2024-07-18:
And the reason this works with pdk .lib files and not elements is that the pdk .lib files are sufficiently detailed as the std cells are small whereas too much detail is lost for Element....

So it is not turtles all the way down...

oharboe, 2024-07-18:
I still get unannoted things.

Untar  [mock-array-element-unannoted.tar.gz](https://github.com/user-attachments/files/16280704/mock-array-element-unannoted.tar.gz)

```
./run-me-mock-array-asap7-base.sh
```

```
OpenSTA 2.5.0 1c7f022cd0 Copyright (c) 2024, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13178, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13211, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13244, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13277, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13310, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13343, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13376, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14772, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14805, timing group from output port.
Warning: /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14838, timing group from output port.
Found 1286 unannotated drivers.
 reset
 ces_0_0/clkload0/Y
 ces_0_0/clkload1/Y
 ces_0_0/clkload10/Y
 ces_0_0/clkload11/Y
 ces_0_0/clkload12/Y
 ces_0_0/clkload13/Y
 ces_0_0/clkload14/Y
 ces_0_0/clkload15/Y
[deleted]
Found 7560 partially unannotated drivers.
 ces_0_0/output259/Y
  ces_0_1/_1282_/A
 ces_0_0/output260/Y
  ces_0_1/_1283_/A
 ces_0_0/output261/Y
  ces_0_1/_1284_/A
 ces_0_0/output262/Y
  ces_0_1/input257/A
 ces_0_0/output263/Y
  ces_0_1/_1285_/A
 ces_0_0/output264/Y
  ces_0_1/_1286_/A
 ces_0_0/output265/Y
  ces_0_1/_1287_/A
 ces_0_0/output394/Y
  ces_0_1/input129/A
 ces_0_0/output395/Y
  ces_0_1/input130/A
 ces_0_0/output396/Y
  ces_0_1/input131/A
 ces_0_0/output397/Y
  ces_0_1/input132/A
 ces_0_0/output398/Y
  ces_0_1/input133/A
 ces_0_0/output399/Y
  ces_0_1/input134/A
 ces_0_0/output400/Y
  ces_0_1/input135/A
 ces_0_0/output401/Y
  ces_0_1/input136/A
 ces_0_0/output402/Y
  ces_0_1/input137/A
 ces_0_0/output403/Y
[... 99 lines cut ...]
  ces_7_6/input128/A
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.98e-01   1.47e-03   4.96e-06   1.99e-01  58.2%
Combinational          3.34e-04   3.67e-04   5.93e-06   7.07e-04   0.2%
Clock                  6.98e-02   7.25e-02   6.48e-07   1.42e-01  41.6%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.68e-01   7.43e-02   1.15e-05   3.42e-01 100.0%
                          78.3%      21.7%       0.0%
Annotated 217414 pin activities.
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.95e-01   0.00e+00   4.89e-06   1.95e-01  57.8%
Combinational          2.98e-14   1.99e-14   5.93e-06   5.93e-06   0.0%
Clock                  6.98e-02   7.25e-02   6.48e-07   1.42e-01  42.2%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.65e-01   7.25e-02   1.15e-05   3.37e-01 100.0%
                          78.5%      21.5%       0.0%
```

maliberty, 2024-07-18:
I'm guessing clock_load cells have floating output so there is nothing to extract in spef.  Are the others similar?

oharboe, 2024-07-18:
Yes... Verilator picks up on it too....

```
%Warning-PINMISSING: /home/oyvind/OpenROAD-flow-scripts/flow/designs/src/mock-array/test_run_dir/MockArray_should_Wiggle_some_wires/MockArrayFinal.v:68344:22: Cell has missing pin: 'Y'
68344 |  INVxp67_ASAP7_75t_R clkload0 (.A(clknet_3_1_0_clock_regs));
      |                      ^~~~~~~~
                     ... For warning description see https://verilator.org/warn/PINMISSING?v=5.012
                     ... Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: /home/oyvind/OpenROAD-flow-scripts/flow/designs/src/mock-array/test_run_dir/MockArray_should_Wiggle_some_wires/MockArrayFinal.v:68345:21: Cell has missing pin: 'Y'
68345 |  BUFx4f_ASAP7_75t_R clkload1 (.A(clknet_3_2_0_clock_regs));
      |                     ^~~~~~~~
%Warning-PINMISSING: /home/oyvind/OpenROAD-flow-scripts/flow/designs/src/mock-array/test_run_dir/MockArray_should_Wiggle_some_wires/MockArrayFinal.v:68346:22: Cell has missing pin: 'Y'
68346 |  INVxp67_ASAP7_75t_R clkload2 (.A(clknet_3_3_0_clock_regs));
```

maliberty, 2024-07-18:
In that case I'm not sure there is anything to fix.

oharboe, 2024-07-18:
> In that case I'm not sure there is anything to fix.

That is one group of warnings, but there are one or two other kinds that I dont understand yet....

maliberty, 2024-07-18:
Do you need help or are you investigating?

oharboe, 2024-07-18:
> Do you need help or are you investigating?

Could you do a run locally and categorize the warnings?

Can we get rid of them somehow by indicating somewhere that this is OK?

It would be satisfying to have no warnings and no wires or pins left unannotated by .vcd and also easier to write a test....

maliberty, 2024-07-18:
report_parasitic_annotation covers spef annotations not vcd.  I'll take a look.

maliberty, 2024-07-18:
The partially annotated drivers come about because the top level abutment.  The extractor doesn't write anything at the top level and sta warns about the partial annotation.  In practice everything is ok but sta doesn't know that.  I'll see if I can convince rcx to write a dummy connection.

oharboe, 2024-07-18:
> The partially annotated drivers come about because the top level abutment. The extractor doesn't write anything at the top level and sta warns about the partial annotation. In practice everything is ok but sta doesn't know that. I'll see if I can convince rcx to write a dummy connection.

👍

oharboe, 2024-07-21:
@maliberty opened #5411 to track the feature request(reduce warnings).

closed 2024-07-21 by oharboe