The-OpenROAD-Project/OpenROAD #5389: Improve navigation with arrow keys in layout viewer

titan73, 2024-07-17:
### Describe the bug

I noticed 2 things while using arrow keys:
- the step is too low
- when keeping an arrow key the scroll bar moves but the layout is not refresh. The refresh is done when the key is released

To improve that would be good to:
- set the scroll step with an option to keep original step by default
- force update of layout view when even if the key is maintained pressed

I plan to give it a try so you can assign the issue to me.
As I'm not a Qt developper, I may ask some question regarding the gui here.

### Expected Behavior

As explained in the description.

### Environment

```shell
It's gui stuff.
```

### To Reproduce

Launch OR as usual.

titan73, 2024-07-17:
It took a look at the code regarding LayoutViewer and LayoutSCroll and I don't see any mention to arrow keys and Z key.

$ grep Qt::Key * -r
src/staGui.cpp:  if (event->key() == Qt::Key_Delete && box_->hasFocus()) {
src/cmdInputWidget.cpp:  if (key == Qt::Key_Down) {
src/cmdInputWidget.cpp:  } else if (key == Qt::Key_Up) {
src/cmdInputWidget.cpp:  if (key == Qt::Key_Enter || key == Qt::Key_Return) {
src/tclCmdInputWidget.cpp:    if (key == Qt::Key_Enter || key == Qt::Key_Return || key == Qt::Key_Escape
src/tclCmdInputWidget.cpp:        || key == Qt::Key_Tab || key == Qt::Key_Backtab) {
src/tclCmdInputWidget.cpp:    bool is_completer_shortcut = has_control && key == Qt::Key_E;  // CTRL+E
src/tclCmdInputWidget.cpp:  if (key == Qt::Key_Enter || key == Qt::Key_Return) {
src/mainWindow.cpp:  if (event->key() == Qt::Key_Escape) {
src/mainWindow.cpp:  } else if (event->key() == Qt::Key_K
 
It there a default behaviour in QScrollArea?
It seems there is but what about Z and SHIFT+Z. It does not seem QSCrollAread supports that?
How is this implemented?

maliberty, 2024-07-17:
Via an action
```
  zoom_in_ = new QAction("Zoom in", this);
  zoom_in_->setShortcut(QString("Z"));
```

maliberty, 2024-07-17:
Arrows are builtin https://doc.qt.io/qt-5/qabstractscrollarea.html#keyPressEvent

titan73, 2024-07-18:
Ok. Thanks for the answers.
I sent a PR for configuring arrow keys scroll step:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/5396

titan73, 2024-07-19:
PR has been merge. Commit https://github.com/The-OpenROAD-Project/OpenROAD/commit/8db034db2173852886c4e463c30c24ac48d97a1b.

I'm trying to see why the layout is not updated while arrow keys are held pressed.
I tried many things but nothing worked.
Also tried a QSCrollArea example. No issue. The area is updated while a key is held pressed. So I guess this is something in OR gui.

There are 2 classes:
   - LayoutViewer which inherits from QWidget
   - LayoutScroll which inherits from QScrollArea

LayoutScroll gets keys and mouse events but it seems this is LayoutViewer that does the rendering.
What is the relation between the 2? And which one does the display?

titan73, 2024-07-19:
I can open a new issue for the remaining problem if needed (and close this one).

maliberty, 2024-07-19:
If you have a very large design then rendering may be slow.  In the upper left we show three dots while rendering.  I did try your PR and it worked for me.  

LayoutViewer has a sub-thread that does the rendering while keeping the GUI live.

titan73, 2024-07-19:
The PR was just to configure the step of the scroll. It was not intended to fix the layout not updated while an arrow is held pressed (of course a big zoom in should have been done prior to the scroll).
You don't have this issue?

titan73, 2024-07-19:
Actually it seems it is a environment problem.

On RedHat 8.9 with openroad (compiled on ubuntu 22.04 under Virtual Box) with .so from ubuntu 22.04 (in a sort of container taken from oss-cad-suite). The issue does not appear.

On Ubuntu 22.04 under Virtual Box, the issue appears.
Note that the issue appears when if a big zoom in is applied. If the zoom in applied is intermediate, then the issue is not present.

So I guess things get weird with the renderer under Ubuntu / Virtual box. The fact the issue happens only with a big zoom in is curious though. So probably a environment issue.

Note that the design I have is not big so I've never seen the three dots yet.

Let's close this issue then.

closed 2024-07-19 by titan73