The-OpenROAD-Project/OpenROAD #5391: DRT: long runtime due to unsolvable DRCs

eder-matheus, 2024-07-17:
### Describe the bug

From https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2142, sky130hs/microwatt has a huge runtime after the changes during synthesis. It happens due to a bad routing resources configuration, leading to many wires assigned to a congested region. DRT takes more than half of the iterations trying to solve 36 violations that look unsolvable.

### Expected Behavior

DRT should give up on hopeless designs.

### Environment

```shell
Git commit: 86e37aa483fe5409095cd4e68ff140f57a19998d
kernel: Linux 6.5.0-1023-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-14595-g86e37aa48
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 88 lines cut ...]
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCLNumber of processor cores: 32
 header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.7iB4sPiEuj
```

### To Reproduce

Test case available here: https://drive.google.com/file/d/1c7lxyFGMgj7dW62qUXk2nnEbXUs1-IAF/view?usp=sharing

gadfort, 2024-07-18:
This is similar to: https://github.com/The-OpenROAD-Project/OpenROAD/issues/5305

gadfort, 2024-07-18:
I think the better solution would be to figure out why DRT is generating so many shorts as the starting point.

eder-matheus, 2024-07-18:
@gadfort For both designs, the problem seems to be the route guides generated by GRT. We were using very low resource adjustments (20%) for these designs, and when I increased it to 40%, both sky130hd/aes and sky130hs/riscv32i finished in less than 20 iterations.

What happens is that when using very low reduction values, GRT assigns too many nets to the same region, which can make DRT struggle to find a clean solution. I'm investigating how to turn GRT smarter about the resources available, in parallel to understanding if it's possible to detect if a design can't be routed without DRCs.

gadfort, 2024-07-20:
@eder-matheus is it possible the original resources are not getting computed correctly? If I don't apply any reductions, I would assume the capacities reported should reflect the number of tracks available.

In the image below, you can see the capacity (20) is reported for the GCell (it's hard to see the outline), but if I count the number of resources it's 16 if I exclude the tracks that are inside the power rails, in this cell, global routing is using 17 resources which might be contributing to the issue.
[image]

It looks like GRT should be derating the power, but this appears to not be the case here.

eder-matheus, 2024-07-21:
@gadfort If I remember correctly, the Capacity shown in the GUI is the first computed (i.e., it considers all the resources calculated before any reduction). The usage would include the reductions from blockages and user-defined adjustments and the routing. I'll double-check if that is the case.

eder-matheus, 2024-07-21:
> @gadfort If I remember correctly, the Capacity shown in the GUI is the first computed (i.e., it considers all the resources calculated before any reduction). The usage would include the reductions from blockages and user-defined adjustments and the routing. I'll double-check if that is the case.

That's how it works. I load the design without running the routing, and it shows what I described: the capacity is the full resources calculated at first, and the reductions from blockages are shown in the usage.

gadfort, 2024-07-21:
@eder-matheus ahh, okay. thanks for checking on that

stefanottili, 2024-07-22:
grt needs to be a lot less optimistic about routing resources … see
https://github.com/The-OpenROAD-Project/OpenROAD/issues/5061

gadfort, 2024-07-22:
@eder-matheus just updating the routing resources per design doesn't seem that viable and overall is a sub-par solution. Now that your CI is down to 9 minutes, how are you going to track the improvements made here?

eder-matheus, 2024-07-22:
@gadfort FYI, I created this issue to track the necessary GRT enhancement regarding routing resource adjustments: https://github.com/The-OpenROAD-Project/OpenROAD/issues/5423.