The-OpenROAD-Project/OpenROAD #5405: GRT/RSZ: Instability in the results of repair_timing command during incremental GRT.

eder-matheus, 2024-07-21:
### Describe the bug

The issue was noticed in the pipelines executed in the PR: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2166. The design asap7/mock-alu failed in one of the pipelines, while it passed in the other. Looking at the log files, the first source of differences is on the repair_timing report during incremental grt. The report diverges between the runs in later iterations of the repair_timing flow.

### Expected Behavior

Both runs should produce the same results.

### Environment

```shell
ORFS PRs pipelines.
```

### To Reproduce

The artifacts for both runs are available here: https://drive.google.com/file/d/10VPaTZKkFswoAbwCnMHA7cOuIaEGW64V/view?usp=sharing

maliberty, 2024-07-21:
This sounds similar to what @osamahammad21 observed between gcp and ucsd

maliberty, 2024-08-13:
@eder-matheus still relevant after the sta thread=1 change?

eder-matheus, 2024-08-13:
Not really, I didn't see it again after setting sta thread to 1.