The-OpenROAD-Project/OpenROAD #5415: Make `info script` return path to script executing

oharboe, 2024-07-22:
### Description

```
$ cat blah.tcl
puts "Script running: [info script]"
$ tclsh blah.tcl 
Script running: blah.tcl
$ openroad -exit blah.tcl
OpenROAD v2.0-14658-gd48987e36 
[deleted]
Script running: 
```

Use case:

Implement the following `source` idiom to remove dependency on pwd, just like `#include "foo.h"` in C++ that includes the foo.h in the same folder as the compilation unit.

```
source [file dirname [info script]]/util.tcl
```

### Suggested Solution

Reduce surprises, make `info script` return same as for `tclsh`

oharboe, 2024-07-22:
Looks like the problem is that `Tcl_Eval` instead of `Tcl_EvalFile` is used.

maliberty, 2024-07-22:
where?

oharboe, 2024-07-22:
> where?

https://github.com/The-OpenROAD-Project/OpenSTA/blob/1c7f022cd0a02ce71d047aa3dbb64e924b6efbd5/app/StaMain.cc#L95

maliberty, 2024-07-22:
So not an openroad issue but an sta one

oharboe, 2024-07-23:
> So not an openroad issue but an sta one

I wouldn't say so. It is the choise of OpenROAD to use "source ..." instead of Tcl_EvalFile(). This choice is encoded into the OpenROAD GUI code:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/0e7dcb1f9100f6667a7d76a39f7d2f0a43749b17/src/Main.cc#L395

https://github.com/The-OpenROAD-Project/OpenROAD/blob/0e7dcb1f9100f6667a7d76a39f7d2f0a43749b17/src/Main.cc#L441

https://github.com/The-OpenROAD-Project/OpenROAD/blob/0e7dcb1f9100f6667a7d76a39f7d2f0a43749b17/src/Main.cc#L401

https://github.com/The-OpenROAD-Project/OpenROAD/blob/0e7dcb1f9100f6667a7d76a39f7d2f0a43749b17/src/Main.cc#L433

oharboe, 2024-07-23:
If #5427 and #5428 are merged, then the remaining failing case for `info script` is when the GUI is enabled, which I think is a somewhat marginal use-case(`info script` is most critical in batch processing) that can be fixed eventually by someone who has a need for it.

```
$ cat blah.tcl
puts "Script running: [info script]"
$ tclsh blah.tcl 
Script running: blah.tcl
$ openroad -gui -exit blah.tcl
OpenROAD v2.0-14658-gd48987e36 
[deleted]
Script running: 
```

Expected output can be seen in non-gui case:

```
$ openroad -exit blah.tcl
OpenROAD v2.0-14711-ga257a8578 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Script running: blah.tcl
```

maliberty, 2024-07-23:
When I asked 'where' you pointed to StaMain.cc but all your subsequent PRs are unrelated to that.  Was that a red herring?

oharboe, 2024-07-23:
Sort of. The fn I referred to was a banal call and error reporting, so I duplicated it and changed to Tcl_EvalFile().

However @gadfort pointed out the the tcl "source" language feature has been monkeypatched to add some logging and verbosity features(not uniquely better on for instance error reporting compared to tcl).

gadfort, 2024-07-23:
@oharboe I think something like: https://github.com/gadfort/OpenSTA/commit/f467add240f120768aeb6478e6b650b8d26945c9 will solve the issue. This ensures the file is set before the "virtual" source is called and reset at the end.

oharboe, 2024-07-24:
> @oharboe I think something like: [gadfort/OpenSTA@f467add](https://github.com/gadfort/OpenSTA/commit/f467add240f120768aeb6478e6b650b8d26945c9) will solve the issue. This ensures the file is set before the "virtual" source is called and reset at the end.

Makes sense... I want to hear on #5436 before forming an opinion... If OpenSTA grows a batch mode where it doesn't introduce features into or change tcl then for my part that solves the issues. My main concern is automated/scripted mode. In the GUI(including REPL console mode), I'm not particularly concerned about the fine details of tcl, there is a human there to intervene when it comes to the userinterface works.

maliberty, 2025-02-02:
Fixed upstream and will be in the next update

closed 2025-02-02 by maliberty

oharboe, 2025-02-02:
> Fixed upstream and will be in the next update

@hovind FYI