The-OpenROAD-Project/OpenROAD #5436: openroad -exit foo.tcl should print out a stacktrace & error message, like tclsh would have

oharboe, 2024-07-23:
### Description

Today `openroad -exit foo.tcl` doesn't print out a stack trace and there are some surprises w.r.t. the tcl `source` command being overriden by OpenSTA

See also https://github.com/The-OpenROAD-Project/OpenROAD/issues/5415

### Suggested Solution

Fewer surprises for the batch(automated) case. OpenSTA should not modify standard tcl behavior.

For the GUI(including console mode REPL), it doesn't matter so much, because there is a user to interpret what is happening and invervene.

maliberty, 2024-07-23:
Are you referring to a stack trace on segv or some tcl stack trace on error?

oharboe, 2024-07-23:
tcl stack trac

oharboe, 2024-07-26:
@tspyrou Tried to investigate some more by moving STA's implementation of "source" into it's own namespace. There are still differences:

```
$ openroad -exit -no_init blah.tcl 
OpenROAD v2.0-14788-g11331a82c 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
blah.tcl
invalid command name "asdf"
    while executing
"::unknown {*}$args"
    (procedure "sta_unknown" line 28)
    invoked from within
"asdf"
    (file "blah.tcl" line 4)
    invoked from within
"source blah.tcl"
$ tclsh blah.tcl 
blah.tcl
invalid command name "asdf"
    while executing
"asdf"
    (file "blah.tcl" line 4)
$ cat blah.tcl 
puts [info script]

asdf
```

oharboe, 2024-07-26:
@tspyrou @maliberty With https://github.com/Pinata-Consulting/OpenSTA/tree/sta-source-namespace and https://github.com/Pinata-Consulting/OpenROAD/tree/sta-source-namespace, I get the expected output:

```
$ cat blah.tcl 
puts [info script]

asdf
$ openroad -exit -no_init blah.tcl 
OpenROAD v2.0-14788-g6c6b363ad 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
blah.tcl
invalid command name "asdf"
    while executing
"asdf"
    (file "blah.tcl" line 4)
$ tclsh blah.tcl 
blah.tcl
invalid command name "asdf"
    while executing
"asdf"
    (file "blah.tcl" line 4)
```

maliberty, 2024-07-26:
Does this also work in you start OR and then do `source blah.tcl` or `source -echo blah.tcl`?

maliberty, 2024-07-26:
Unknown is intended as an extension point in TCL.  Virtually every EDA does uses it to avoid the problem :
```
# Bus signal names like foo[2] or bar[31:0] use brackets that
# look like "eval" to TCL. Catch the numeric "function" with the
# namespace's unknown handler and return the value instead of an error.
```

I don't see that as an issue.

oharboe, 2024-07-26:
> Unknown is intended as an extension point in TCL. Virtually every EDA does uses it to avoid the problem :
>
> ```
> # Bus signal names like foo[2] or bar[31:0] use brackets that
> # look like "eval" to [...]

Not as such, but I had to disable it to avoid some sort of crash. Didn't investigate more at the time, just trying to get a better sense of scope of niggling little differences between stock Tcl and OpenSTA Tcl tweaks...