The-OpenROAD-Project/OpenROAD #5440: rsz: Rebuffering wire delay model can be better

povik, 2024-07-24:
### Describe the bug

Take a look at `Rebuffer.cc` at `rebufferBottomUp`:
```
   double layer_res, layer_cap;
    bnet_wire->wireRC(corner, resizer_, layer_res, layer_cap);
    double wire_res = wire_length * layer_res;
    double wire_cap = wire_length * layer_cap;
    double wire_delay = wire_res * wire_cap;
    BufferedNetPtr z = make_shared<BufferedNet>(
        BufferedNetType::wire, wire_end, wire_layer, p, corner, resizer_);
    // account for wire load
    z->setCapacitance(p->cap() + wire_cap);
    z->setRequiredPath(req_path);
    // account for wire delay
    z->setRequiredDelay(p->requiredDelay() + wire_delay);
```
the wire delay is taken to be `R*C` where `R` and `C` refer to the wire itself, there's no taking account of the capacitance downstream of the wire segment. This has an effect when different buffering options are evaluated later on within the same file.

### Expected Behavior

The delay is modeled with `R*C/2 + R*C_downstream` or similar

### Environment

```shell
N/A
```

### To Reproduce

N/A

precisionmoon, 2024-07-31:
I think you mean function RepairSetup::addWireAndBuffer().  Yes, if we had a distributed RC or a pie RC, we can compute Elmore delay using downstream cap.   The code seems to be focusing on a pre-route case without extracted RC.  If we have the full distributed RC available, we can cut the RC network at the desired buffer location to do more accurate estimation and para update.  This will require significant code change though.

povik, 2024-08-07:
> I think you mean function RepairSetup::addWireAndBuffer().

Yes, that's what I meant, thanks.

> Yes, if we had a distributed RC or a pie RC, we can compute Elmore delay using downstream cap. The code seems to be focusing on a pre-route case without extracted RC.

At that place within the rebuffering code, we can use the same RC estimates which `estimate_parasitics -placement` is using. E.g. to implement the `R*C/2 + R*C_downstream` model:

```
diff --git a/src/rsz/src/Rebuffer.cc b/src/rsz/src/Rebuffer.cc
index a7ed71319..dd8310f0f 100644
--- a/src/rsz/src/Rebuffer.cc
+++ b/src/rsz/src/Rebuffer.cc
@@ -330,7 +330,7 @@ BufferedNetSeq RepairSetup::addWireAndBuffer(const BufferedNetSeq& Z,
     bnet_wire->wireRC(corner, resizer_, layer_res, layer_cap);
     double wire_res = wire_length * layer_res;
     double wire_cap = wire_length * layer_cap;
-    double wire_delay = wire_res * wire_cap;
+    double wire_delay = wire_res * (wire_cap / 2 + p->cap());
     BufferedNetPtr z = make_shared<BufferedNet>(
         BufferedNetType::wire, wire_end, wire_layer, p, corner, resizer_);
     // account for wire load
```

PR #6637 (2025-04-02): rsz: Revamp rebuffering
Improve the rebuffering code:

 * replace "slack penalization" with separate timing-oriented and area-oriented passes over the buffer tree; this should allow better area recovery on non-critical sections of the buffer tree, especially if the tree is large
 * make slight adjustments to the delay model
 * restructure some of the code

Fixes #5440, #5441
closed 2025-04-02 by maliberty