The-OpenROAD-Project/OpenROAD #5449: Can't build OpenROAD Flow Scripts Locally, but can without flag --local

dzemildzigal, 2024-07-25:
### Describe the bug

Hello, I've successfully built OpenROAD Flow Scripts using ./build_openroad.sh and made the Docker container. 

I want to **specifically build it locally** using the --local flag but I'm failing in the start. I've ran the ./etc/DependencyInstaller.sh script beforehand. 

I'm erroring out almost immediately and here is the output:
`(core-v-mini-mcu) foo@bar:~/.../OpenROAD-flow-scripts$ ./build_openroad.sh --local
[INFO FLW-0028] Compiling with 4 threads.
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --local
tee: build_openroad.log: Permission denied
[INFO FLW-0002] Updating git submodules.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/home/jemo/Desktop/AnilProjekat/x-heep/flow/OpenROAD-flow-scripts/tools/yosys'
[  0%] Building kernel/driver.o
[  0%] Building kernel/register.o
[  1%] Building kernel/rtlil.o
[  1%] Building kernel/log.o
In file included from kernel/register.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/driver.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/rtlil.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/log.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
make: *** [Makefile:729: kernel/driver.o] Error 1
make: *** Waiting for unfinished jobs....
1 error generated.
1 error generated.
make: *** [Makefile:729: kernel/log.o] Error 1
make: *** [Makefile:729: kernel/register.o] Error 1
1 error generated.
make: *** [Makefile:729: kernel/rtlil.o] Error 1
make: Leaving directory '/home/jemo/Desktop/AnilProjekat/x-heep/flow/OpenROAD-flow-scripts/tools/yosys'
`
Building without the --local flag passes and all dependencies should be satisfied in any case.

### Expected Behavior

I expect the build to pass successfully, not fail immediately.

### Environment

```shell
foo@bar:/.../OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-44-generic
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
[ 11%] Creating directories for 'googletest-populate'
[ 22%] Performing download step (download, verify and extract) for 'googletest-populate'
-- Downloading...
[... 982 lines cut ...]
  /usr/local/share/cmake-3.24/Modules/FetchContent.cmake:1958 (FetchContent_Populate)
  third-party/gtest/CMakeLists.txt:8 (FetchContent_MakeAvailable)

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-14726-g72ee0f9c4
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Configuring incomplete, errors occurred!
```

### Additional Context

Ubuntu 22.04.4 LTS, 64-bit, 
Intel® Core™ i5-7300HQ CPU @ 2.50GHz (4 core, 8 threads)
16GB RAM

dzemildzigal, 2024-07-25:
Same thing happens when running with sudo (permission denied errors are solved then):

sudo ./build_openroad.sh --local
[sudo] password for jemo: 
[INFO FLW-0028] Compiling with 4 threads.
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --local
[INFO FLW-0002] Updating git submodules.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/home/jemo/Desktop/AnilProjekat/x-heep/flow/OpenROAD-flow-scripts/tools/yosys'
[  0%] Building kernel/driver.o
[  0%] Building kernel/register.o
[  1%] Building kernel/rtlil.o
In file included from [  1%] Building kernel/log.o
kernel/driver.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/register.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/rtlil.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/log.cc:20:
In file included from ./kernel/yosys.h:42:
./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
make: *** [Makefile:729: kernel/driver.o] Error 1
make: *** Waiting for unfinished jobs....
1 error generated.
make: *** [Makefile:729: kernel/log.o] Error 1
1 error generated.
make: *** [Makefile:729: kernel/register.o] Error 1
1 error generated.
make: *** [Makefile:729: kernel/rtlil.o] Error 1
make: Leaving directory '/home/jemo/Desktop/AnilProjekat/x-heep/flow/OpenROAD-flow-scripts/tools/yosys'

eder-matheus, 2024-07-25:
Some developers had a similar error, solved in Ubuntu with `sudo apt install g++-12`. Could you try it?

dzemildzigal, 2024-07-26:
> Some developers had a similar error, solved in Ubuntu with `sudo apt install g++-12`. Could you try it?

Hi, thank you, this helped. Installing g++ 12 solved and am able to build it locally. I am not going to close the thread until I fully built it from the sources yet as this process takes a bit of time so we'll see soon enough.

dzemildzigal, 2024-07-26:
Build passed. Closing the issue now.

closed 2024-07-26 by dzemildzigal