The-OpenROAD-Project/OpenROAD #5463: Add find by name to CTS viewer

maliberty, 2024-07-27:
### Description

To track:

From @precisionmoon `We want to find a register by name in the main GUI and see the matching sink in the clock tree viewer.`

### Suggested Solution

From @gadfort `It should be fairly easy to implement since you can piggy-back on the inspect connection to emit a second call to a sink in the viewer that can highlight it.`

maliberty, 2024-07-27:
@gadfort do you want to tackle this or should I pass to @AcKoucher ?

gadfort, 2024-07-27:
@maliberty if @AcKoucher has time that would probably be best right now.

maliberty, 2024-08-08:
@AcKoucher please help @LucasYuki with this as a first project.

AcKoucher, 2024-08-08:
Should we have a "Find" option exclusively for the ClockTreeViewer or this sink highlighting should be triggered by the main window "Find"?

maliberty, 2024-08-08:
I think the existing find mechanism is fine.  The CTS viewer should just listen for selection in a similar fashion to the layout viewer.

LucasYuki, 2024-08-19:
While talking to @arthurjolo and @AcKoucher we have the following question:

The existing find mechanism can find multiple instances in one search.
In this case, how is the CTS viewer expected to behave?

What if we implement in the inspector a button, or an option with the right-click menu, to "find in the CTS viewer"?

maliberty, 2024-08-19:
Is there a reason we couldn't support multi-select in the CTS viewer?  When you get a select you zoom to the bounds of the selected object(s).

PR #5665 (2024-09-17): gui: Add find by name to CTS viewer 
Closes #5463.

Changes:
- Makes the find dialog search instances in the CTS Viewer. (Implements #5463)
- Find dialog focuses on the find text box when the popup opens

If instances are found, the CTS Viewer zooms to fit all found objects. 
When the design has multiple clock signals, the current tree changes to where the instances are found.
closed 2024-09-17 by maliberty