The-OpenROAD-Project/OpenROAD #5469: remove gui_floorplan bogus `[ERROR GRT-0010] Instance ctrl.state.out\[0\]$_DFF_P_ is not placed.`

oharboe, 2024-07-29:
### Description

```
$ make floorplan gui_floorplan
QSocketNotifier: Can only be used with threads started with QThread
OpenROAD v2.0-14802-gc7e334587 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR GRT-0010] Instance ctrl.state.out\[0\]$_DFF_P_ is not placed.
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0066] Heat map "Estimated Congestion (RUDY)" has not been populated with data.
```

### Suggested Solution

Remove bogus error.

maliberty, 2024-07-29:
Presumably you have the estimated congestion heatmap on.  There is no way to display it with unplaced instances.  I don't see this as a bogus error - just turn it off.

oharboe, 2024-07-29:
This happens because OpenROAD remembers settings across all runs...

Not a truly great userexperience, but there are certainly bigger fish to fry.

closed 2024-07-29 by oharboe

maliberty, 2024-07-29:
We could not remember the heatmaps but that also has its downsides.

oharboe, 2024-07-29:
Yes... I think weit

oharboe, 2024-07-29:
I didnt explicitly turn it on. Normally in a GUI, unavailable options are ghosted silently.

maliberty, 2024-07-29:
You did in a previous run, presumably when it was available.

oharboe, 2024-07-29:
And then I restarted the GUI, loaded another design, and now it is not.

My expectation is silently ghosting unavailable options, that is not what OpenROAD does here.