The-OpenROAD-Project/OpenROAD #5471: Make the ruler flexible

titan73, 2024-07-29:
### Description

With the current ruler, making a measurement requires to snap to 2 edges. Unfortunately in some cases and particularly in debug mode where we deal with temporary object, there is not way to do a measurement and it's not practical at all. That would be really convenient to chooses arbitrary locations. This is possible with commercial tool such as Innovus and Calibre.

### Suggested Solution

To be defines. In Calibre, there is a option to choose arbitrary location. With Innovus, it's mixed. We can choose arbitrary locations and when we're close to an edge, the tool snap to it.

maliberty, 2024-07-29:
You can do that with ctrl-click

titan73, 2024-07-30:
Just tried. Great. Thanks!

closed 2024-07-30 by titan73