The-OpenROAD-Project/OpenROAD #5475: megaboom global placement takes 3 hours

oharboe, 2024-07-30:
### Description

This is the megaboom test-case, in case it is useful...

untar https://drive.google.com/file/d/187o-cj4575guAmgYxNS4v7XYCVtmDcAd/view?usp=sharing

```
./run-me-BoomTile-asap7-base.sh
```

```
$ grep GPL_ vars-BoomTile-asap7-base.sh 
export GPL_ROUTABILITY_DRIVEN="0"
export GPL_TIMING_DRIVEN="1"
```

### Suggested Solution

Make it faster...

oharboe, 2024-07-30:
Trying to disable timing driven placement here to see what happens... https://github.com/The-OpenROAD-Project/megaboom/pull/71

gudeh, 2024-07-30:
Hi @oharboe, I noticed you disabled routability driven, it should not be a bottleneck anymore because now we use RUDY instead of fastroute. Although, I would expect timing driven to be slow because of rsz calls to repair design.

I am currently working on some changes in timing driven, but these changes should not reduce runtime, at least for now.