The-OpenROAD-Project/OpenROAD #5486: Error when calling detailed_route

titan73, 2024-07-31:
### Describe the bug

Here is the error meesage:

Error: test.tcl, 25 vector::_M_range_check: __n (which is 4) >= this->size() (which is 0)

### Expected Behavior

Should succeeded

### Environment

```shell
Latest git.
```

### To Reproduce

openroad test.tcl with test case provided.

[pin_access2.zip](https://github.com/user-attachments/files/16438851/pin_access2.zip)

maliberty, 2024-07-31:
@osamahammad21 this is a very small test case so hopefully this is quick to resolve.

osamahammad21, 2024-07-31:
@titan73 the verilog code doesn't make any sense.
```
module test(a, b, z);
  input a;
  wire a;
  input b;
  wire b;
  output z;
  wire z;
  STD_CELL _00_ (
    .A(flops[1]),
    .B(in_ports[1]),
    .Z(out_ports[1])
  );
endmodule
```
 There is no nets to route that's why DRT breaks and GRT does not do anything.

maliberty, 2024-07-31:
@osamahammad21 it would be nice to give a clear error in that case and stop early.

titan73, 2024-07-31:
Ah indeed. Sorry. I missed some renaming when making the test case. This should have been:

module test(a, b, z);
  input a;
  wire a;
  input b;
  wire b;
  output z;
  wire z;
  STD_CELL _00_ (
    .A(a),
    .B(b),
    .Z(z)
  );
endmodule

It's curious read_verilog didn't complain at all.

osamahammad21, 2024-07-31:
> @osamahammad21 it would be nice to give a clear error in that case and stop early.

the issue originates with DRT's global router and I don't think it is a priority at this point. maybe GRT should have an error or warning if there are no nets to route!

osamahammad21, 2024-08-04:
I am closing this issue since there is no reply for 4 days. If needed, please reopen.

titan73, 2024-08-08:
> I am closing this issue since there is no reply for 4 days. If needed, please reopen.

@osamahammad21 I was in holidays. There was no question for me but for @maliberty (see just below)

> @titan73 you might as well see it ending with one offgrid violation that cannot be solved. this is because metal1's half width is not divisible by the manufacturing grid. @maliberty not sure if there should be an error or a warning for this case?

How did you see that offgrid violations? I didn't see anything and I'd like to take a look.

> > @osamahammad21 it would be nice to give a clear error in that case and stop early.
> 
> the issue originates with DRT's global router and I don't think it is a priority at this point. maybe GRT should have an error or warning if there are no nets to route!

I'll try to add a check in GRT. We can keep either this issue open or reopen it (I can't to it) and assigned it to me. As you prefer.

titan73, 2024-08-09:
There was nets but nothing to route indeed. I just noticed the top ports are iterms. I would have expected bterms.

nb nets = 6
name a, bterms 0, iterms 1
name b, bterms 0, iterms 1
name flops[1], bterms 1, iterms 0
name in_ports[1], bterms 1, iterms 0
name out_ports[1], bterms 1, iterms 0
name z, bterms 0, iterms 1

maliberty, 2024-08-09:
Top level ports are bterms.  What is generating this output?

titan73, 2024-08-09:
It's info printed when debugging:

```
  bool has_routable_nets = false;
  printf(">>>>>>>>>> nb nets = %d\n", db_->getChip()->getBlock()->getNets().size());
  for( auto net : db_->getChip()->getBlock()->getNets()) {
    printf("name %s, bterms %d, iterms %d\n", net->getName().c_str(), net->getITerms().size(), net->getBTerms().size());
    if (net->getITerms().size() + net->getBTerms().size() > 1) {
      has_routable_nets = true;
      break;
    }
  }

```

titan73, 2024-08-09:
Ah my bad I inverted the values. The top ports are bterms and standard cells pins are iterms.

titan73, 2024-08-16:
PR has been merged. Thanks for the support.

closed 2024-08-16 by titan73