The-OpenROAD-Project/OpenROAD #5530: cmake try to install dependency googletest/archive/refs/tags/v1.13.0.zip

titan73, 2024-08-08:
### Describe the bug

As I understand, etc/DependencyInstaller.sh is supposed to do this job.

### Expected Behavior

Should not deal with dependencies installation.

### Environment

```shell
Latest git.
```

### To Reproduce

cd build && cmake ..

### Relevant log output

```shell
CMake Error at googletest-subbuild/googletest-populate-prefix/src/googletest-populate-stamp/download-googletest-populate.cmake:170 (message):
  Each download failed!
    error: downloading 'https://github.com/google/googletest/archive/refs/tags/v1.13.0.zip' failed
          status_code: 6
          status_string: "Couldn't resolve host name"
          log:
          --- LOG BEGIN ---
          getaddrinfo(3) failed for github.com:443
  Could not resolve host: github.com
```

maliberty, 2024-08-08:
Looks like you have a dns problem or a network issue as there is nothing we can do if github.com can't be resolved.  I don't see this as an OR issue.  Is it a transient problem?

titan73, 2024-08-08:
No it's not transient. I can't install dependencies on my laptop from my company network so I use my mobile phone to avoid the firewall. I generally do it when running etc/DependencyInstaller.sh then switch back to company network for all the rest.
I can do it also when running cmake but what is the reason some dependencies are install during cmake. Should be done with etc/DependencyInstaller.sh? Is there a particular reason?

maliberty, 2024-08-08:
@QuantamHD I believe you set this up so perhaps you can answer.

QuantamHD, 2024-08-08:
The reason is that relying on the system Gtest results in a lot of incompatibility, and the prefered installation path is to dynamically load in the project the way we've done so. 

@titan73 We could probably offer an override var to the URL so you could set it to some file:/// path on your local machine.  

Would that be sufficient?

gadfort, 2024-08-08:
@titan73 could you disable the test builds with `-DENABLE_TESTS=OFF` that should prevent the GoogleTest stuff from triggering?

titan73, 2024-08-09:
@gadfort Yes indeed. This prevent the issue.

titan73, 2024-08-09:
@QuantamHD Maybe my question was not clear. I understand the needs to use custom version of tools rather than using the system version. The question was about why it's done during cmake rather the DependencyInstaller.sh

For example, in DependencyInstaller.sh we use a custom version of cmake (3.24.2) rather than the system one.
Why not doing the same for gtest? Is there a reason not to do it in DependencyInstaller.sh?

titan73, 2024-08-22:
PR has been merged. Closing.

closed 2024-08-22 by titan73