The-OpenROAD-Project/OpenROAD #5537: cts segfault

oharboe, 2024-08-09:
### Describe the bug

Untar https://drive.google.com/file/d/1aN1Trih2vu1Q7n69BsYstnKDVXtASz9P/view?usp=sharing

```
./run-me-BoomTile-asap7-base.sh
```

Output:

```
[INFO CTS-0017]     Max level of the clock tree: 11.
Signal 11 received
Stack trace:
 0# 0x000062A30C346FF3 in openroad
 1# 0x000078014D042990 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbITerm::getMTerm() const in openroad
 3# odb::dbITerm::isInputSignal(bool) in openroad
 4# cts::TritonCTS::getFirstInput(odb::dbInst*) const in openroad
 5# 0x000062A30E38F34C in openroad
 6# 0x000062A30E38DF2E in openroad
 7# cts::TritonCTS::computeIdealOutputCaps(cts::Clock&) in openroad
 8# cts::TritonCTS::writeDummyLoadsToDb(cts::Clock&, std::unordered_set<odb::dbInst*, std::hash<odb::dbInst*>, std::equal_to<odb::dbInst*>, std::allocator<odb::dbInst*> >&) in openroad
 9# cts::TritonCTS::writeDataToDb() in openroad
10# cts::TritonCTS::runTritonCts() in openroad
11# 0x000062A30CA74599 in openroad
12# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
13# 0x0000780151997C43 in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
16# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
17# ord::tclAppInit(Tcl_Interp*) in openroad
18# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
19# main in openroad
20# 0x000078014D028150 in /lib/x86_64-linux-gnu/libc.so.6
21# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
22# _start in openroad
./run-me-BoomTile-asap7-base.sh: line 7: 689656 Segmentation fault      (core dumped) openroad -no_init ${SCRIPTS_DIR}/cts.tcl
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-15103-gc7aa07725
```

### To Reproduce

See above

oharboe, 2024-08-09:
@jeffng-or FYI with https://github.com/The-OpenROAD-Project/megaboom/commit/742df07506f1b2894cd31089422e99b138eaec36

oharboe, 2024-08-09:
@maliberty Is there some magic option we can add to stop this crash from happening while it is being fixed?

oharboe, 2024-08-09:
The crash has to be fixed in CTS of course(possibly replaced with an error message), but the root cause could be earlier.

```
$ grep -i error bazel-bin/logs/asap7/BoomTile/base/3_*.log
$ grep -i warning bazel-bin/logs/asap7/BoomTile/base/3_*.log
$ grep -i error bazel-bin/logs/asap7/BoomTile/base/2_*.log
$ grep -i warning bazel-bin/logs/asap7/BoomTile/base/2_*.log
bazel-bin/logs/asap7/BoomTile/base/2_1_floorplan.log:Warning: There are 148 input ports missing set_input_delay.
bazel-bin/logs/asap7/BoomTile/base/2_1_floorplan.log:Warning: There are 238 output ports missing set_output_delay.
bazel-bin/logs/asap7/BoomTile/base/2_1_floorplan.log:Warning: There are 3784 unconstrained endpoints.
bazel-bin/logs/asap7/BoomTile/base/2_1_floorplan.log:[WARNING IFP-0028] Core area lower left (1.000, 1.000) snapped to (1.026, 1.080).
bazel-bin/logs/asap7/BoomTile/base/2_6_floorplan_pdn.log:[WARNING PDN-0195] Removing 5 via(s) between M2 and M5 at (22.9260 um, 468.1830 um) for VSS
bazel-bin/logs/asap7/BoomTile/base/2_6_floorplan_pdn.log:[WARNING PDN-0195] Removing 5 via(s) between M2 and M5 at (28.3260 um, 468.1830 um) for VSS
bazel-bin/logs/asap7/BoomTile/base/2_6_floorplan_pdn.log:[WARNING PDN-0195] Removing 5 via(s) between M2 and M5 at (33.7260 um, 468.1830 um) for VSS
bazel-bin/logs/asap7/BoomTile/base/2_6_floorplan_pdn.log:[WARNING PDN-0195] Removing 5 via(s) between M2 and M5 at (39.1260 um, 468.1830 um) for VSS
[lost more, but same kind]
```

precisionmoon, 2024-08-09:
I am looking into this with high priority.   Will send some update by EOB today.

precisionmoon, 2024-08-09:
The fatal happens during the dummy load insertion.   You can avoid the fatal by using -dont_use_dummy_load option for now.

precisionmoon, 2024-08-13:
We have identified the root cause for the fatal and are evaluating a solution (PR  5560).    We want to merge this by Aug 16 after all the validation completes.

oharboe, 2024-08-13:
@precisionmoon Thanks! I think we need this fix + sta upgrade, before we can upgrade megaboom to use latest ORFS.

closed 2024-08-13 by maliberty