The-OpenROAD-Project/OpenROAD #5542: Add grouping to report_power for instances, just like for top level

oharboe, 2024-08-09:
### Description

It would be useful to have the grouping for an instance(a Verilog module) that I get for the top level:

"report_power" (top level):

```
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.70e+00   7.59e-01   6.48e-06   2.46e+00  59.9%
Combinational          8.28e-01   6.59e-01   4.74e-06   1.49e+00  36.2%
Clock                  8.49e-02   7.69e-02   7.85e-07   1.62e-01   3.9%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.62e+00   1.49e+00   1.20e-05   4.11e+00 100.0%
                          63.6%      36.4%       0.0%
```

For an instance, in my case MockArray Element(which is  a Verilog module & a macro), I get:

"report_power -instances ces_7_7"?

```
   Internal  Switching    Leakage      Total
      Power      Power      Power      Power (Watts)
--------------------------------------------
   2.97e-02   1.56e-02   1.96e-07   4.53e-02 ces_7_7
```

See https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2228 for how `report_power -instances ces_7_7` is used.

### Suggested Solution

Add grouping like there is at top level

maliberty, 2024-08-09:
@tspyrou for Cherry

tspyrou, 2024-08-09:
@oharboe are you planning to run this hierarchically or do it flat like with the final sta?
We can add the groups to write_timing_model and then use them when a macro is reported for sure.
I had been thinking that you would do this flat. In either case its a good enhancement. I am wondering about the priority.

oharboe, 2024-08-10:
see mock-array "make simulation power" for use case

oharboe, 2024-08-10:
Medium to low pri.