The-OpenROAD-Project/OpenROAD #5554: [WARNING DRT-0225] CFG_LBCSYNCMODE 10 pin not visited, fall back to feedthrough mode

stefanottili, 2024-08-12:
### Describe the bug

DRT is spewing DRT-0225 messages util it hits the limit, so there is no way of knowing whether this is for every single pin or not.
This is from the ICCAD2004 testcase, TSMC 180nm, artisan stdcell lib.
Routes cleanly.

### Expected Behavior

What is this warning all about ?
The pins look normal to me, all of them are on track, easy to route to.

### Environment

```shell
Mac M1
OpenROAD v2.0-15115-g20ac1be86 (Aug 11th 2024)
Features included (+) or not (-): +Charts +GPU +GUI +Python
```

### To Reproduce

Please get lef/def, risc2.or and Makefile from
https://github.com/The-OpenROAD-Project/OpenROAD/issues/5548

### Relevant log output

```shell
379 [INFO DRT-0028]   Complete metal6.
 380 [WARNING DRT-0225] CFG_LBCSYNCMODE 10 pin not visited, fall back to feedthrough mode.
 381 [WARNING DRT-0225] CFG_MEMZEROFIRST 2 pin not visited, fall back to feedthrough mode.
...
1379 [WARNING DRT-0225] lx1/CE0/macd/macfile/n358 4 pin not visited, fall back to feedthrough mode.
1380 [WARNING DRT-0225] message limit reached, this message will no longer print
```

maliberty, 2024-08-12:
@osamahammad21 please do a quick triage.  I'm not sure if this is a drt issue or a grt one.

osamahammad21, 2024-09-10:
Feedthrough mode means that we are using the pins as nodes to connect the net guides. Forexample, if we have a pin that has shapes on layers M1 M2 and M3, and 2 guides reaching in/out that pin on M1 and M3, then we use the pin as a connection between the M1 guide and M3 guide.

titan73, 2024-09-10:
So this mode prevents to use a guide in M2 right? And the pin not visited is the shape in M2? It's a kind of optimization?

osamahammad21, 2024-09-10:
I wouldn't say it's an optimization. Before going through detailed routing we have to verify the ocrrectness of the global routes first. This is what we are doing here. We are traversing the guides for each net to make sure that all pins are covered. If not, then we issue an error and do not go into detailed routing.
Ideally the guides should be all connected together without the need of using pins as feedthrough nodes in graph. But some global route solutions consider the pins as part of net guides graph. In that case we warn the user.
In all cases we don't attempt to fix the global routes.

osamahammad21, 2024-09-10:
Ok I see the problem now, the global router is using metal1 as the bottom routing layer.
As far as a user's concern, you should set the min routing layer to be M2 as the cells are placed on M1.
However, this has other implications for the tool developers:
- GRT again miscalculates M1 resources. @eder-matheus https://github.com/The-OpenROAD-Project/OpenROAD/issues/5696
- in DRT, we should either get rid of the warning or make it more informative. In guides processing, we do two iterations of graph traversals. In the first, we consider the m1 guides as via guides only and pins as in/terminals. In the second, we consider M1 guides as segment guides and pins as inout(feedthrough) terminals. In that case, the first iteration fails because the M1 guides are considered as vias, so they are not connected.  (Although the number of pins visited should be 1 only leaving 2 not visited pins!)

closed 2025-01-07 by osamahammad21