The-OpenROAD-Project/OpenROAD #5557: wrong RUDY map for centered floorplan

stefanottili, 2024-08-12:
### Describe the bug

Follow up of https://github.com/The-OpenROAD-Project/OpenROAD/issues/5300.

Since Aug 12th https://github.com/The-OpenROAD-Project/OpenROAD/pull/5549 the GRT resource map is correct.
OR can GPL/DPL/GRT/DRT a centered floorplan.

RUDY resource calculation/display is wrong.

FARADAY_ICCAD04Bench/RISC2
Without DRT this takes 25 seconds to run.

### Expected Behavior

RUDY matching GRT map.

### Environment

```shell
M1 Mac
OpenROAD v2.0-15128-g136639014 (Aug 12th 2024)
Features included (+) or not (-): +Charts +GPU +GUI +Python
```

### To Reproduce

[centered_noram.tgz](https://github.com/user-attachments/files/16589684/centered_noram.tgz)

### Screenshots

GRT
[image]
RUDY
[image]

stefanottili, 2025-01-27:
I wanted to see how the updated RUDY code behaves on this test case.
But as of Jan 27 OpenRoad seems to be stuck in global route 'Extra Run for hard benchmark' for > 15min ...
Getting to a global routed database took 25 seconds in August 2024.
```
[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
metal1     Horizontal     381441        146055          61.71%
metal2     Vertical       381441        379042          0.63%
metal3     Horizontal     381441        379042          0.63%
metal4     Vertical       306075        304150          0.63%
metal5     Horizontal     178080        176960          0.63%
metal6     Vertical       178080        176960          0.63%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
```

stefanottili, 2025-01-27:
Unfortunately the log still doesn't have any runtime info in it, but after what felt like > 15 min, global route finished.

The good news is that this ticked can be closed.
RUDY not only correlates with the global route map, but it display's the centered floorpan correctly, yay.

The bad news is that as of Jan 27 the placer creates an unroutable placement.
[image]
rudy
[image]
global route
[image]

closed 2025-01-27 by stefanottili

gudeh, 2025-01-27:
Hi @stefanottili, isn't this issue about the centered floorplan?

stefanottili, 2025-01-27:
That issue seems to be fixed as you can see in the attached RUDY map, that's why I closed this issue.

But it revealed another issue:
what used place/global route in 25 sec, is now unroutable due to a bad placement.

stefanottili, 2025-01-27:
running with "first quadrant" floorplan, placement finishes in no time with a routable floorpan.

So even thought the RUDY map is displayed correctly, gpl using RUDY seems to have an issue with a centered floorpan.

gudeh, 2025-01-27:
I made no modifications to RUDY. Interesting how the issue stopped happening.

stefanottili, 2025-01-27:
What changed in gpl ? In August 2024 this placed/routed just fine.

gudeh, 2025-01-27:
There could have been floorplan and logic synthesis modifications, if this designs happens to run this stages prior to gpl.

If you are running timing_driven mode, we also have the influence of rsz and sta modifications.

Do you run any of this for this design?

gudeh, 2025-01-27:
Also, there could have been modifications to the GUI.

stefanottili, 2025-01-27:
Same netlist, same floorplan.
That Rudy is displayed correctly might be just a gui update.
Something changed in gpl so that it’s not producing a routable placement any more.

gudeh, 2025-01-27:
are you running with timing_driven mode? Could you share the logs you have for gpl?

stefanottili, 2025-01-27:
Not timing driven, lef/def + openroad command file in the tar file, it takes 25 to run this.

stefanottili, 2025-01-27:
```
openroad -gui risc2.or | tee log
qt.qpa.fonts: Populating font family aliases took 54 ms. Replace uses of missing font family "Monospace" with one that exists to avoid this cost. 
OpenROAD v2.0-18638-gf17b89150 
2025-01-27 12:10:10.103 openroad[97734:275935644] +[IMKClient subclass]: chose IMKClient_Modern
2025-01-27 12:10:10.103 openroad[97734:275935644] +[IMKInputSession subclass]: chose IMKInputSession_Modern
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING ODB-0220] WARNING (LEFPARS-2003): BUSBITCHARS is a required statement on LEF file with version 5.5 and earlier.
Without BUSBITCHARS defined, the LEF file is technically incorrect.
Refer the LEF/DEF 5.5 or earlier Language Referece manual on how to define this statement. See file lef/risc2.lef.gz at line 22827.

[WARNING ODB-0220] WARNING (LEFPARS-2004): DIVIDERCHAR is a required statementon LEF file with version 5.5 and earlier.
Without DIVIDECHAR defined, the LEF file is technically incorrect.
Refer the LEF/DEF 5.5 or earlier Language Referece manual on how to define this statement. See file lef/risc2.lef.gz at line 22827.

[INFO ODB-0227] LEF file: lef/risc2.lef.gz, created 13 layers, 16 vias, 147 library cells
[INFO ODB-0127] Reading DEF file: def/risc2.noram.def.gz
[INFO ODB-0128] Design: frisc
[INFO ODB-0130]     Created 629 pins.
[INFO ODB-0131]     Created 32615 components and 195545 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 65230 connections.
[INFO ODB-0133]     Created 34034 nets and 128024 connections.
[INFO ODB-0134] Finished DEF file: def/risc2.noram.def.gz
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: (  0.400  3.600 ) um
[INFO GPL-0004] CoreBBox: ( -479.600 -478.400 ) ( 480.000 479.200 ) um
[INFO GPL-0006] NumInstances:             32615
[INFO GPL-0007] NumPlaceInstances:        32615
[INFO GPL-0008] NumFixedInstances:            0
[INFO GPL-0009] NumDummyInstances:            0
[INFO GPL-0010] NumNets:                  34032
[INFO GPL-0011] NumPins:                 127666
[INFO GPL-0012] DieBBox:  ( -479.600 -479.600 ) ( 480.000 480.000 ) um
[INFO GPL-0013] CoreBBox: ( -479.600 -478.400 ) ( 480.000 479.200 ) um
[INFO GPL-0016] CoreArea:            918912.960 um^2
[INFO GPL-0017] NonPlaceInstsArea:        0.000 um^2
[INFO GPL-0018] PlaceInstsArea:      521791.200 um^2
[INFO GPL-0019] Util:                    56.784 %
[INFO GPL-0020] StdInstsArea:        521791.200 um^2
[INFO GPL-0021] MacroInstsArea:           0.000 um^2
[InitialPlace]  Iter: 1 CG residual: 0.02409817 HPWL: 20887486040
[InitialPlace]  Iter: 2 CG residual: 0.01398059 HPWL: 722098063
[InitialPlace]  Iter: 3 CG residual: 0.00848638 HPWL: 694950880
[InitialPlace]  Iter: 4 CG residual: 0.00403918 HPWL: 679288861
[InitialPlace]  Iter: 5 CG residual: 0.00240668 HPWL: 668348594
[InitialPlace]  Iter: 6 CG residual: 0.00097734 HPWL: 661660560
[InitialPlace]  Iter: 7 CG residual: 0.00105022 HPWL: 654290412
[InitialPlace]  Iter: 8 CG residual: 0.00082364 HPWL: 651881683
[InitialPlace]  Iter: 9 CG residual: 0.00073686 HPWL: 648391348
[InitialPlace]  Iter: 10 CG residual: 0.00048074 HPWL: 646219410
[InitialPlace]  Iter: 11 CG residual: 0.00061198 HPWL: 643120807
[InitialPlace]  Iter: 12 CG residual: 0.00056104 HPWL: 641215428
[InitialPlace]  Iter: 13 CG residual: 0.00045857 HPWL: 639051713
[InitialPlace]  Iter: 14 CG residual: 0.00040547 HPWL: 637868695
[InitialPlace]  Iter: 15 CG residual: 0.00024440 HPWL: 635911931
[InitialPlace]  Iter: 16 CG residual: 0.00026331 HPWL: 634795794
[InitialPlace]  Iter: 17 CG residual: 0.00017507 HPWL: 633353986
[InitialPlace]  Iter: 18 CG residual: 0.00015163 HPWL: 632451702
[InitialPlace]  Iter: 19 CG residual: 0.00013986 HPWL: 631271314
[InitialPlace]  Iter: 20 CG residual: 0.00013997 HPWL: 630648790
[INFO GPL-0031] FillerInit:NumGCells:     40603
[INFO GPL-0032] FillerInit:NumGNets:      34032
[INFO GPL-0033] FillerInit:NumGPins:     127666
[INFO GPL-0023] TargetDensity:            0.700
[INFO GPL-0024] AvrgPlaceInstArea:       15.999 um^2
[INFO GPL-0025] IdealBinArea:            22.855 um^2
[INFO GPL-0026] IdealBinCnt:              40206
[INFO GPL-0027] TotalBinArea:        918912.960 um^2
[... 127 lines cut ...]
[NesterovSolve] Iter:  300 overflow: 0.587 HPWL: 1031684905
[NesterovSolve] Iter:  310 overflow: 0.543 HPWL: 1046124153
[NesterovSolve] Iter:  320 overflow: 0.497 HPWL: 1065639906
[NesterovSolve] Iter:  330 overflow: 0.451 HPWL: 1070665152
[NesterovSolve] Iter:  340 overflow: 0.406 HPWL: 1075479013
[NesterovSolve] Iter:  350 overflow: 0.357 HPWL: 1076455636
[NesterovSolve] Iter:  360 overflow: 0.321 HPWL: 1078246173
[NesterovSolve] Iter:  370 overflow: 0.287 HPWL: 1078272789
[NesterovSolve] Iter:  380 overflow: 0.254 HPWL: 1080662500
[NesterovSolve] Iter:  390 overflow: 0.225 HPWL: 1081844233
[NesterovSolve] Iter:  400 overflow: 0.195 HPWL: 1083311449
[NesterovSolve] Iter:  410 overflow: 0.170 HPWL: 1084903666
[NesterovSolve] Iter:  420 overflow: 0.145 HPWL: 1086919338
[NesterovSolve] Iter:  430 overflow: 0.123 HPWL: 1088680829
[NesterovSolve] Iter:  440 overflow: 0.104 HPWL: 1090177559
[NesterovSolve] Finished with Overflow: 0.099042
Placement Analysis
---------------------------------
total displacement    3613512.4 u
average displacement      110.8 u
max displacement          736.0 u
original HPWL         1091240.1 u
legalized HPWL        4943157.6 u
delta HPWL                  353 %
```