The-OpenROAD-Project/OpenROAD #5574: GUI crash

oharboe, 2024-08-13:
### Describe the bug

untar https://drive.google.com/file/d/17VPUOTE0FLmW9ei9kHb5LVDs7GBLhFxj/view?usp=sharing

```
./run-me-swerv_wrapper-asap7-base.sh
```

Output, crash:

```
$ ./run-me-swerv_wrapper-asap7-base.sh 
QSocketNotifier: Can only be used with threads started with QThread
OpenROAD v2.0-15137-g35b37708a 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[deleted]
[WARNING GRT-0026] Missing route to pin swerv/ifu/aln/_7546_/Y.
[WARNING GRT-0026] Missing route to pin swerv/ifu/aln/_7549_/Y.
[WARNING GRT-0026] Missing route to pin swerv/ifu/aln/_7568_/Y.
[WARNING GRT-0026] Missing route to pin swerv/ifu/aln/_7569_/Y.
[WARNING GRT-0026] message limit reached, this message will no longer print
Signal 11 received
Stack trace:
 0# 0x00005C98AA86F463 in openroad
 1# 0x0000794146242990 in /lib/x86_64-linux-gnu/libc.so.6
 2# grt::FastRouteCore::getEdgeCapacity(int, int, int, int, int) in openroad
 3# grt::FastRouteCore::initBlockedIntervals(std::vector<int, std::allocator<int> >&) in openroad
 4# grt::GlobalRouter::applyAdjustments(int, int) in openroad
 5# grt::GlobalRouter::initFastRoute(int, int) in openroad
 6# grt::Rudy::Rudy(odb::dbBlock*, grt::GlobalRouter*) in openroad
 7# grt::GlobalRouter::getRudy() in openroad
 8# grt::RUDYDataSource::populateXYGrid() in openroad
 9# gui::HeatMapDataSource::setupMap() in openroad
10# gui::HeatMapDataSource::ensureMap() in openroad
11# gui::HeatMapRenderer::drawObjects(gui::Painter&) in openroad
12# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in openroad
13# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
14# gui::RenderThread::run() in openroad
15# 0x00007941472DAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
16# 0x0000794146297B5A in /lib/x86_64-linux-gnu/libc.so.6
17# 0x00007941463285FC in /lib/x86_64-linux-gnu/libc.so.6
./run-me-swerv_wrapper-asap7-base.sh: line 7: 2930147 Segmentation fault      (core dumped) openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-15137-g35b37708a
```

### To Reproduce

See above

oharboe, 2024-08-13:
The test-case that crashes originates from from https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2243

```
bazel run swerv_wrapper_route `pwd`/build
`pwd`/build/make gui_issue
```

maliberty, 2024-08-13:
@gudeh this looks more like a grt/rudy issue than a GUI one so please take a look.

oharboe, 2024-08-14:
`make issue` does specify the `-no_init` option... so this should happen whether or not I enabled RUDY in a previous run of OpenROAD GUI or not...

```
$ cat run-me-swerv_wrapper-asap7-base.sh 
#!/usr/bin/env bash
source vars-swerv_wrapper-asap7-base.sh
if [[ ! -z ${GDB+x} ]]; then
    gdb --args openroad -no_init ${SCRIPTS_DIR}/gui.tcl
else
    openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
fi
```

oharboe, 2024-08-14:
@gudeh I had some troubles building OpenROAD, but was just able to.

The problem appears to be gone in 35b37708a v2.0-15137-g35b37708a, has it been fixed to your knowledge?

oharboe, 2024-08-14:
aha! Then that could be the issue. @maliberty I talked about it in the past.

Load one design, turn on RUDY.

Load some other design where RUDY shouldn't be turned on(e.g. showing floorplan where it is not available), crash/strange errors.

oharboe, 2024-08-14:
I think this is just more of the same:

```
[WARNING GRT-0026] Missing route to pin clkload0/A.
/usr/include/c++/13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = int; _Alloc = std::allocator<int>; reference = int&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x000063B1F646D9C3 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x0000745E5E442990 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x0000745E5E8E219F in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# grt::FastRouteCore::initBlockedIntervals(std::vector<int, std::allocator<int> >&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# grt::GlobalRouter::applyAdjustments(int, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# grt::GlobalRouter::initFastRoute(int, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# grt::Rudy::Rudy(odb::dbBlock*, grt::GlobalRouter*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# grt::GlobalRouter::getRudy() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# grt::RUDYDataSource::populateXYGrid() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# gui::HeatMapDataSource::setupMap() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# gui::HeatMapDataSource::ensureMap() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# gui::HeatMapRenderer::drawObjects(gui::Painter&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# gui::RenderThread::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# 0x0000745E5F4DAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
19# 0x0000745E5E497B5A in /lib/x86_64-linux-gnu/libc.so.6
20# 0x0000745E5E5285FC in /lib/x86_64-linux-gnu/libc.so.6
make: *** [/home/oyvind/OpenROAD-flow-scripts/flow/Makefile:1061: gui_5_route.odb] Aborted (core dumped)
```

gudeh, 2024-08-14:
I was able to avoid the crash by adding a simple if condition to check for valid bounds before accessing the vector. I will create a PR.

gadfort, 2024-08-14:
@maliberty I do as well. We just need to ensure the tools they are connected to dont segfault on launch which seems reasonable anyways.

oharboe, 2024-08-14:
> I like having the heatmap visibility persist but if most others don't I can be persuaded.

Can we make it a silent no-op when it is not relevant?

Today there are million warnings if I have it left on from studying something in place and I load something up in floorplan...

gudeh, 2024-08-16:
Crash solved with PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/5578. Further modifications will be made with issue https://github.com/The-OpenROAD-Project/OpenROAD/issues/5594

closed 2024-08-16 by gudeh