The-OpenROAD-Project/OpenROAD #5577: Resizing Problems

AcKoucher, 2024-08-14:
### Description

Currently, RepairDesign uses a strategy of resizing cells to a target slew in order to perform Post Placement Optimization. With the current methodology, the Resizer ends up "fixing" things that are not actually violations of any rule leading to over-sizing.

Note: The API that resizes to the target slew is also used in RepairSetup and RepairHold.

### Suggested Solution

(Some) Current problems:
  - [ ] Cells are resized regardless of violations (No DRC, Setup/Hold violations)
  - [ ] FFs are being touched.
  - [ ] Special cells are also touched. They should be checked with the cell footprint.
  - [ ] Clock buffers are used for data network.

Add fanout-of-4 (FO4) Rule
  - [ ] Gates that don't respect FO4 should be resized.
  - [ ] Gates with cap ratio that don't violate FO4 shouldn't be further sized

maliberty, 2024-11-05:
@povik you will take over the remaining tasks

maliberty, 2025-08-12:
@povik how much of phases 2 & 3 remains?  We should make sub-issues for anything yet to do as this is rather too large an item to track.

povik, 2025-08-13:
From phase 2:

> Do sequential sizing based on area increase (nominal buffer area threshold)

is missing

From phase 3, all is missing.

povik, 2025-08-13:
I'll discuss "Do sequential sizing based on area increase" with Cho.

Ad phase 3:

* https://github.com/The-OpenROAD-Project/OpenROAD/issues/8013
* https://github.com/The-OpenROAD-Project/OpenROAD/issues/8014

povik, 2025-08-13:
That's right, though we still want to adjust when in the flow we invoke it.