The-OpenROAD-Project/OpenROAD #5579: recent updates to sta broke build on MacOS: cmake can't find sta/TCL_READLINE_INCLUDE

stefanottili, 2024-08-14:
### Describe the bug

recent changes to sta broke the build on MacOS

etc/DependencyInstaller.sh doens't install cudd when using homebrew
easy to fix by using https://github.com/mht208/homebrew-formal/blob/master/cudd.rb
brew install ~/Downloads/cudd.rb

But then I ran into
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
OpenROAD-flow-scripts/tools/OpenROAD/src/sta/TCL_READLINE_INCLUDE
   used as include directory in directory OpenROAD-flow-scripts/tools/OpenROAD/src/sta
```

### Expected Behavior

be able to build on MacOS --local

### Environment

```shell
M1 Mac
git checkout as of 14th Aug 2024
```

### To Reproduce

./build_openroad.sh --local --latest

### Relevant log output

```shell
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
OpenROAD-flow-scripts/tools/OpenROAD/src/sta/TCL_READLINE_INCLUDE
   used as include directory in directory OpenROAD-flow-scripts/tools/OpenROAD/src/sta
```

maliberty, 2024-08-14:
@eder-matheus please comment as you use a mac

eder-matheus, 2024-08-14:
I have this error in cmake, but when I run make, it compiles to the end. It was already solved on the latest sta: https://github.com/parallaxsw/OpenSTA/commit/754dc6255ed37a450dc77030dcdc48cb48c1bef0

maliberty, 2024-08-14:
Should be fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/5581 - please reopen if not.

closed 2024-08-14 by maliberty

stefanottili, 2024-08-14:
works again, thanks

Still, DependencyInstaller.sh needs to be updated to install cudd on Mac ...

maliberty, 2024-08-14:
Would you or @eder-matheus look into this as I have no mac to try.