The-OpenROAD-Project/OpenROAD #5583: Pin Overlapping Issue with Power Switch Cells in Sky130 Library

bingyuew, 2024-08-15:
### Description

The power switch described here: [src/pdn/test/sky130_power_switch/power_switch.lef](https://github.com/The-OpenROAD-Project/OpenROAD/blob/bf8a6a7a1c86511cbcfd56556743d1af4c0f2ac0/src/pdn/test/sky130_power_switch/power_switch.lef) has an overlap between the VDDG and SLEEP_OUT pins. 

[image]

This overlap will cause an error during pin access in the routing stage. Furthermore, the test script located at [[src/pdn/test/sky130_power_switch/floorplan.def](https://github.com/The-OpenROAD-Project/OpenROAD/blob/bf8a6a7a1c86511cbcfd56556743d1af4c0f2ac0/src/pdn/test/power_switch_daisy.tcl)](https://github.com/The-OpenROAD-Project/OpenROAD/blob/bf8a6a7a1c86511cbcfd56556743d1af4c0f2ac0/src/pdn/test/power_switch_daisy.tcl) appears to be incorrect due to this issue.

maliberty, 2024-08-15:
@gadfort any comments?

gadfort, 2024-08-15:
@maliberty the pin is overlapping (I can fix that although it doesn't matter for PDNs sake). I'm not sure what @bingyuew means by the tcl script is invalid.

maliberty, 2024-08-15:
The script mentioned is a unit test that only runs pdn.  It would be problematic to routing if we were running it.  Is that the concern?

vikrg23, 2024-08-16:
Hi @gadfort , I work with @bingyuew  , sorry the tcl script is not the issue, we misunderstood. Just the overlap in the power switch LEF is the issue. Thanks.

maliberty, 2024-08-16:
In that case it looks like this is resolved by https://github.com/The-OpenROAD-Project/OpenROAD/pull/5586

closed 2024-08-16 by maliberty