The-OpenROAD-Project/OpenROAD #5584: Non-fatal floorplan error

oharboe, 2024-08-15:
See https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2249

Also, to create an issue using a *local* ORFS, you can do:

```
bazel run BoomTile_floorplan_deps `pwd`/build
build/make do-floorplan
```

this quickly shows:

```
Running floorplan.tcl, stage 2_1_floorplan
[INFO ORD-0030] Using 20 thread(s).
[INFO ODB-0227] LEF file: external/bazel-orfs~~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: external/bazel-orfs~~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells
[WARNING ODB-0177] error: undefined via (VIA45_1_2_58_58) referenced
[WARNING ODB-0177] error: undefined via (VIA45_1_2_58_58) referenced
[INFO ODB-0227] LEF file: results/asap7/ebtb_128x40/base/ebtb_128x40.lef, created 2 vias, 1 library cells
```

Leave it running to see if it gets further, but already you can open a new shell an create the first issue:

```
$ build/make floorplan_issue
mkdir -p .//objects/asap7/BoomTile/base
bash: line 1: V: command not found
make: [external/bazel-orfs~~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/Makefile:395: versions.txt] Error 127 (ignored)
external/bazel-orfs~~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/util/makeIssue.sh floorplan
bash: line 1: external/bazel-orfs~~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/util/makeIssue.sh: No such file or directory
make: *** [external/bazel-orfs~~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/util/utils.mk:98: floorplan_issue] Error 127
$ . ~/OpenROAD-flow-scripts/env.sh 
OPENROAD: /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD
$ build/make floorplan_issue
[deleted output]
```

I like to strip down the generated archive more if I can:

```
mkdir foo
cd foo
tar --strip-components=1 -xzvf ../build/floorplan_BoomTile_asap7_base_2024-08-15_06-40.tar.gz
# Put *only* openroad in the path, the rest should come from the archive when testing, *not* your ORFS install, so *don't* run ". env.sh"
export PATH=$PATH:/home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-15055-gdb7d56e4d 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0227] LEF file: /home/oyvind/megaboom/foo/platforms/asap7/lef/asap7_tech_1x_201209.lef, created 24 layers, 9 vias
[INFO ODB-0227] LEF file: /home/oyvind/megaboom/foo/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef, created 212 library cells
[WARNING ODB-0177] error: undefined via (VIA45_1_2_58_58) referenced
[WARNING ODB-0177] error: undefined via (VIA45_1_2_58_58) referenced
[rest deleted]
```

Now trim down what is in this folder to bring down the size of the test case and check that the test still fails.
 
I can reproduce with 40mByte instead of 800mByte...

Still too big for github, so I upload to google drive and create a shareable link:

https://drive.google.com/file/d/1893fWw-vCFPkQY1swQF68vDPSkCdcIay/view?usp=sharing

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/bazel-orfs/issues/99#issuecomment-2290597986_

oharboe, 2024-08-15:
@jeffng-or FYI

maliberty, 2024-08-15:
We want to generate a full set of problems rather than stop on the first one (similar to c++ compilation).  So we give warnings and then error at the end.

oharboe, 2024-08-15:
@maliberty Makes sense now that I know this, maybe the output could make it more clear? No specific suggestions.

maliberty, 2024-08-15:
Is ebtb_128x40.lef new?  It seems like this would always have been a problem.

oharboe, 2024-08-15:
> Is ebtb_128x40.lef new? It seems like this would always have been a problem.

Yes, generated same [version](https://github.com/The-OpenROAD-Project/megaboom/blob/cf6c7eecc80183e3526e2d821d7f9013acbe880e/MODULE.bazel#L27) as megaboom

oharboe, 2024-08-15:
@maliberty Helpful? 

[pdn_ebtb_128x40_asap7_base_2024-08-15_16-05.tar.gz](https://github.com/user-attachments/files/16625995/pdn_ebtb_128x40_asap7_base_2024-08-15_16-05.tar.gz)

maliberty, 2024-08-15:
@gadfort not properly in what way?  I do see it in the .lef but we only read vias from the tech lef today.

maliberty, 2024-08-15:
@gadfort I see we read vias if they have a via rule but these vias don't.  I'm guessing this relates to your recent pdn change.  Thoughts?

gadfort, 2024-08-15:
Since that via is a block via, the abstract lef would have to write it out as boxes instead of referencing it by name

gadfort, 2024-08-15:
I don't immediately see how it was included in the lef from https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/odb/src/lefout/lefout.cpp (it looks like the shapes should have been extracted)

maliberty, 2024-08-15:
From the LEF manual
```
If a LEF MACRO block needs its own locally defined VIA definition that is not already defined,
it should use a unique via name. One way to do so is to use the MACRO name in the VIA
definition name. For example, MACRO my_block1 might have a unique via named VIA
my_block1_via1 inside the same LEF file.
```
so it sounds like there is an intention that LEF could have extra vias.

maliberty, 2024-08-15:
I presume it can from lefout::writeBlockVia

gadfort, 2024-08-15:
we don't do the name ubiquitination which would be a good idea. Do macro lef vias get added to the techvias during lefin?

maliberty, 2024-08-15:
No that's the issue here:
```
void lefin::via(lefiVia* via, dbTechNonDefaultRule* rule)
{
  if (!_create_tech && !via->hasViaRule()) {
    return;
  }
```
The question is should we add them as dbTechVia - it seems we should and just remove the if.  Agree?

maliberty, 2024-08-15:
(We hit this earlier with viarule in https://github.com/The-OpenROAD-Project/OpenROAD/commit/0728be75e10f0cfbb8d18a3e4ddc1300a9eff291)

gadfort, 2024-08-15:
@maliberty I think they should just be added to the Tech (if there is an issue in the future they could tagged as foreign vias).  If we do add them I think it makes sense to follow the LEF guidance about making the names unique

PR #5589 (2024-08-15): odb: handle via in macro LEFs
In lefin, read all vias.  In lefout, prefix vias with the block name.

Fixes #5584
commit c0e6ce9f (2024-08-15): odb: handle via in macro LEFs

In lefin, read all vias.  In lefout, prefix vias with the block name.

Fixes #5584
closed 2024-08-15 by maliberty