The-OpenROAD-Project/OpenROAD #5585: [ERROR PDN-0233] Failed to generate full power grid.

oharboe, 2024-08-15:
### Describe the bug

@jeffng-or discovered this problem in https://github.com/The-OpenROAD-Project/megaboom/pull/84

To reproduce:

untar https://drive.google.com/file/d/1prc-4jBVJvfxHdZ4mr-dZSZU2xrSBGwJ/view?usp=sharing

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-15055-gdb7d56e4d 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: top
[deleted]
[WARNING PDN-0232] CORE_macro_grid_1 - dcache/data/array_0_0_ext does not contain any shapes or vias.
[deleted]
[WARNING PDN-0232] CORE_macro_grid_1 - dcache/meta_1/tag_array_ext does not contain any shapes or vias.
[ERROR PDN-0233] Failed to generate full power grid.
Error: pdn.tcl, 5 PDN-0233
openroad> 
```

### Expected Behavior

No error

### Environment

```shell
OpenROAD v2.0-15055-gdb7d56e4d
```

### To Reproduce

See above

gadfort, 2024-08-15:
@oharboe as far as I can tell, the macros all go to M6 and your top level of PDN straps end at M6 so there is no way to connect to the macros.

oharboe, 2024-08-15:
@jeffng-or Can you fix this? mock-array has a fix for this.

@gadfort Correct?

gadfort, 2024-08-15:
@oharboe I'm not sure what you mean by "correct?"

oharboe, 2024-08-15:
Does mock array do this correctly?

maliberty, 2024-08-15:
I thought this problem appeared in your upgrade.  Is there something new as it doesn't seem pdn has changed.

gadfort, 2024-08-15:
@maliberty @oharboe I had assumed it was related to the via change, but this seems unrelated to that. You would know if the macros have changed their powergrid since that is what would be different here.

maliberty, 2024-08-15:
I don't know that there was an OR change to explain a prior false negative.

jeffng-or, 2024-08-16:
It looks like mock-array uses $(PLATFORM_DIR)/openRoad/pdn/BLOCKS_grid_strategy.tcl instead of $(PLATFORM_DIR)/openRoad/pdn/grid_strategy-M1-M2-M5-M6.tcl.

When I hack the testcase to use BLOCKS_grid_strategy.tcl, pdngen completes with the following warnings:

[WARNING PDN-0239] Core ring shape falls outside the die bounds.
[INFO PDN-0001] Inserting grid: top
[WARNING PDN-0195] Removing 1 via(s) between M5 and M6 at (-0.3300 um, 2157.0560 um) for VDD
[WARNING PDN-0195] Removing 1 via(s) between M5 and M6 at (0.2700 um, 2156.4360 um) for VSS
[WARNING PDN-0110] No via inserted between M5 and M6 at (71.8900, 892.9670) - (71.8980, 893.0320) on VDD
[WARNING PDN-0110] No via inserted between M5 and M6 at (475.7940, 590.4240) - (475.8180, 590.6320) on VDD
[WARNING PDN-0110] No via inserted between M5 and M6 at (963.8580, 2029.0790) - (963.9030, 2029.1920) on VDD
[WARNING PDN-0195] Removing 1 via(s) between M5 and M6 at (2156.4360 um, 2156.4360 um) for VSS
[WARNING PDN-0195] Removing 1 via(s) between M5 and M6 at (2157.0360 um, 2157.0560 um) for VDD

I ended up calling check_power_grid on the resulting ODB, but the job gets killed while checking VSS.

I need to figure out how to pipe this change through on MegaBoom to see how it works in practice

oharboe, 2024-08-17:
@jeffng-or Yes: $PLATFORM_DIR/openRoad/pdn/BLOCKS_grid_strategy.tcl  and $PLATFORM_DIR/openRoad/pdn/BLOCK_grid_strategy.tcl  are strategies that go together for the mock-array use-case where macros are combined at the top level.

gadfort, 2024-08-17:
@jeffng-or I think you will need to recreate the megaboom subclocks to fix their power grids and then try to rebuild megaboom

oharboe, 2024-08-17:
subblocks?

gadfort, 2024-08-17:
@oharboe all the macros in megaboom need to be rebuilt with the correct power grid.

oharboe, 2024-08-17:
ok. we do that every time we upgrade orfs.