The-OpenROAD-Project/OpenROAD #5590: Post-GRT, resizer fails with cryptic error

donn, 2024-08-15:
### Describe the bug

Running post-GRT timing optimizations fails with a cryptic error and later a stack dump (see log)
[repro.partab.tgz](https://github.com/user-attachments/files/16627652/repro.partab.tgz)
, but only in the event that the database has been written after global routing, OpenROAD exited out of, then the database is re-read.

If GRT and resizing happens in the same script, this works fine. However, this may be a waste of compute for some designs where GRT takes a non-trivial amount of time.

### Expected Behavior

Whatever data global routing saves should be in Odb/more actionable error

### Environment

```shell
Git commit: 14b1cbdf0c3160dd34ff79cf413d1f66f9ec4eac
kernel: Darwin 23.5.0
os: macOS 14.5
cmake version 3.29.2
CMake Warning (dev) at src/sta/CMakeLists.txt:32 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  For compatibility with older versions of CMake, option is clearing the
  normal variable 'USE_TCL_READLINE'.
This warning is for project developers.  Use -Wno-dev to suppress it.

Number of processor cores: 24
CMake Warning:
  Manually-specified variables were not used by the project:

    VERBOSE

-- The CXX compiler identification is Clang 16.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/c1ib07kj7baid3s3wf5m98hcpbvgs8mk-clang-wrapper-16.0.6/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15221-g14b1cbdf0
-- System name: Darwin
-- Compiler: Clang 16.0.6
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is Clang 16.0.6
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/c1ib07kj7baid3s3wf5m98hcpbvgs8mk-clang-wrapper-16.0.6/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
[... 138 lines cut ...]
[WARNING STA-0441] set_input_delay relative to a clock defined on the same port/pin not allowed.
[INFO]: Cap load range: 0.5 : 1.0
[INFO]: Input transition range: 1 : 1.19
[INFO]: Setting derate factor to: 15.0 %
[INFO] Setting RC values…
[INFO RSZ-0094] Found 557 endpoints with setup violations.
[INFO RSZ-0099] Repairing 557 out of 557 (100.00%) violating endpoints...
[ERROR GRT-0226] Type2 ripup not type L.
Error: rsz_timing_postgrt.tcl, 36 GRT-0226
Signal 11 received
Stack trace:
 0# handler(int) in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
 1# _sigtramp in /usr/lib/system/libsystem_platform.dylib
 2# std::__1::__tree<std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>, std::__1::__map_value_compare<sta::Instance*, std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>, std::__1::less<sta::Instance*>, true>, std::__1::allocator<std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>>>::destroy(std::__1::__tree_node<std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>, void*>*) in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
 3# std::__1::__tree<std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>, std::__1::__map_value_compare<sta::Instance*, std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>, std::__1::less<sta::Instance*>, true>, std::__1::allocator<std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>>>::destroy(std::__1::__tree_node<std::__1::__value_type<sta::Instance*, std::__1::tuple<sta::LibertyPort*, sta::LibertyPort*>>, void*>*) in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
 4# rsz::Resizer::~Resizer() in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
 5# rsz::Resizer::~Resizer() in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
 6# ord::OpenRoad::~OpenRoad() in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
 7# __cxa_finalize_ranges in /usr/lib/system/libsystem_c.dylib
 8# exit in /usr/lib/system/libsystem_c.dylib
 9# ord::tclAppInit(Tcl_Interp*) in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
10# Tcl_MainEx in /nix/store/7cpj5zdlz28y6ipg2h4cw6qagvqycm0i-tcl-8.6.13/lib/libtcl8.6.dylib
11# main in /nix/store/x1n09k7mal26ln8z7npr0nhd8qq21846-openroad/bin/.openroad-wrapped
./run_bad.sh: line 17: 37378 Segmentation fault: 11  openroad -exit -no_splash -metrics ./or_metrics_out.json ./scripts/openroad/rsz_timing_postgrt.tcl
```

maliberty, 2024-08-15:
`[ERROR GRT-0226] Type2 ripup not type L.` suggest the problem is in the fastroute core.

eder-matheus, 2024-08-22:
The current incremental grt relies on running grt and rsz in the same script. This is because the internal structures that grt uses to estimate the parasitics and to define what nets are already routed are not present in the database.

To fix this issue, we would need to save the route segments of each net in addition to the route guides. This way rsz will get the correct data about connectivity between the nets.

maliberty, 2024-08-22:
> If GRT and resizing happens in the same script, this works fine. However, this may be a waste of compute for some designs where GRT takes a non-trivial amount of time.

Most likely any fix is going to involve re-running both.  Storing all the extra data in the db seems like a lot of overhead for a small benefit.  

The error message should be better regardless.

donn, 2024-08-23:
fyi @kareefardi

kareefardi, 2024-08-25:
> Most likely any fix is going to involve re-running both. Storing all the extra data in the db seems like a lot of overhead for a small benefit.
>
> The error message should be better regardless.

How much is a lot of overhead? Also, how much would rerunning GRT modify existing GRT data (guides only I suppose) in the db?

maliberty, 2024-08-25:
You are storing the routing of each net twice - once in guides and again in segments.  I don't have a quotative value but that isn't trivial.

I'm not sure what the value of this use model is.  Would you explain further how this is a benefit.

closed 2024-08-26 by eder-matheus

kareefardi, 2024-09-09:
@eder-matheus @maliberty I just noticed the commands `read_global_route_segments` and `write_global_route_segments`. Would rsz still need an iteration of GRT even with these commands

maliberty, 2024-09-09:
Those were added for a contest and I don't know we'll keep them long term.

eder-matheus, 2024-09-17:
@kareefardi This PR will make it possible to estimate parasitics from the guides stored in odb: https://github.com/The-OpenROAD-Project/OpenROAD/pull/5758

PR #5638 (2024-08-26): Grt: Avoiding using estimate_parasitics without running global route first
The GRT guides do not have enough information to run estimate parasitics. Global route needs to be run for a correct calculation.

- Adding a warning message when global routing is not found for nets and removing the call to the function to load the routing from the ODB guides.
- Modifying the way to check for the existence of routing in ANT, adding warning message when detailed routing is not found and modifying the error message in case there is no detailed or global routing.

Fixes #5590