The-OpenROAD-Project/OpenROAD #5600: Gui gets stuck after link_design

AcKoucher, 2024-08-16:
### Describe the bug

After running the test case provided in #5340, after link_design finishes running and sta tries to report, the gui doesn't show nothing in scripting and keeps "running".

### Expected Behavior

The report should be visible in Scripting and the GUI should be unpaused.

### Environment

```shell
HEAD of Master
```

### To Reproduce

`openraod -gui test.tcl` with the test case from the description.

titan73, 2024-10-22:
Any news on this issue?

AcKoucher, 2024-10-23:
@titan73 I haven't had the time to look at this.

maliberty, 2025-03-25:
There is something odd about this design.  There are huge numbers of high fanout nets.  I suspect you have some sort of connectivity problem in this nets.  Some examples which repeat for many different values of the integers:

bp_processor/coh_resp_hor_link_li_2__0__data__43_ has 8586 pins
Net bp_processor/coh_req_ver_link_li[507] has 5716 pins
Net bp_processor/cc/lce_cmd_link_li[1054] has 14280 pins

shifengzhicheng, 2025-03-26:
I run the following script:
```tcl
# Setting lef files
set tech_dir "/scripts/MacroPlacement/Enablements/NanGate45"
set tech_lef "${tech_dir}/lef/NangateOpenCellLibrary.tech.lef"
set std_lef "${tech_dir}/lef/NangateOpenCellLibrary.macro.mod.lef"

set lefs "
    ./fakeram45_256x16.lef
"
# Setting lib files
set libs "
    ${tech_dir}/lib/NangateOpenCellLibrary_typical.lib \
    ./fakeram45_256x16.lib \
"

read_lef $tech_lef
read_lef $std_lef

foreach lef_file ${lefs} {
  read_lef $lef_file
}

foreach lib_file ${libs} {
  read_liberty $lib_file
}

puts "read ariane_fp.def"

read_verilog ./ariane.v
link_design ariane
read_sdc ./ariane.sdc
read_def -floorplan_initialize ./ariane_fp.def

# Liberty units are fF,kOhm
set_layer_rc -layer metal1 -resistance 5.4286e-03 -capacitance 7.41819E-02
set_layer_rc -layer metal2 -resistance 3.5714e-03 -capacitance 6.74606E-02
set_layer_rc -layer metal3 -resistance 3.5714e-03 -capacitance 8.88758E-02
set_layer_rc -layer metal4 -resistance 1.5000e-03 -capacitance 1.07121E-01
set_layer_rc -layer metal5 -resistance 1.5000e-03 -capacitance 1.08964E-01
set_layer_rc -layer metal6 -resistance 1.5000e-03 -capacitance 1.02044E-01
set_layer_rc -layer metal7 -resistance 1.8750e-04 -capacitance 1.10436E-01
set_layer_rc -layer metal8 -resistance 1.8750e-04 -capacitance 9.69714E-02
# No calibration data available for metal9 and metal10
#set_layer_rc -layer metal9 -resistance 3.7500e-05 -capacitance 3.6864e-02
#set_layer_rc -layer metal10 -resistance 3.7500e-05 -capacitance 2.8042e-02

set_wire_rc -signal -layer metal3
set_wire_rc -clock  -layer metal5
```
The process gets stuck at the `link_design ariane` step when I start OpenROAD with the `-gui` option. However, if I start OpenROAD from the shell without the `-gui` option, it does not hang at that step.

As a temporary workaround, I execute the `link_design` command in shell and then invoke `gui::show` to bypass the issue. Nonetheless, I believe this is a bug that needs to be resolved.

For debugging purposes, I've uploaded my run script and related code here:
https://github.com/shifengzhicheng/hierarchical-mpl/tree/5e4a74494c7337b4ad79356a0baa481ec926d190/log/Issue_20250326

maliberty, 2025-03-26:
What version of OR are you testing with?  "HEAD of Master" is not descriptive at it changes constantly.

maliberty, 2025-03-26:
That commits is from last Nov.  Please try with the current head.

titan73, 2025-03-26:
I retested with be6d1c688c219e29569c2fffa8cf60b95d9fbd12 for March 18th and the issue is still here.
Without GUI (all print iterations and durations printed in terminal console)
    - 100 print iterations take 1 ms
    - link_design takes 3 mn
    - 100 print iterations take 1 ms
With GUI:
    - 100 print iterations take 6 ms (all printed in terminal console with final duration)
    -  Only "Iteration 0" printed in GUI console
    - Then a popup appears about "OpenROAD not responding", clicking "wait" makes it disappear but the popu reappears 2 s later
    - link_desing actually finish with a duration of 5 mn (printed in the terminal console)
    - "Iteration 1" appears in GUI console
    - Then nothing more happens

maliberty, 2025-03-26:
I found it takes 19 minutes for me w/o the GUI.  I don't think machine speed would explain it taking 3min for you.  Are we running the same data?

titan73, 2025-03-26:
Yes indeed and my setup is not particularly powerful (VM using 8 cores on a HP EliteBook G9). 
I used test case https://github.com/user-attachments/files/16179674/slow.zip from https://github.com/The-OpenROAD-Project/OpenROAD/issues/5340.