The-OpenROAD-Project/OpenROAD #5603: CUDD not directly linked against OpenROAD

donn, 2024-08-16:
### Describe the bug

Newer versions of OpenSTA require CUDD and links it as follows:

```cmake
if (CUDD_LIB)
  message(STATUS "CUDD library: ${CUDD_LIB}")
  get_filename_component(CUDD_LIB_DIR "${CUDD_LIB}" PATH)
  get_filename_component(CUDD_LIB_PARENT1 "${CUDD_LIB_DIR}" PATH)
  find_file(CUDD_HEADER cudd.h
    PATHS ${CUDD_LIB_PARENT1} ${CUDD_LIB_PARENT1}/include ${CUDD_LIB_PARENT1}/include/cudd)
  if (CUDD_HEADER)
    get_filename_component(CUDD_INCLUDE "${CUDD_HEADER}" PATH)
    message(STATUS "CUDD header: ${CUDD_HEADER}")
  else()
    message(STATUS "CUDD header: not found")
  endif()
else()
  message(STATUS "CUDD library: not found")
endif()

// …

target_link_libraries(OpenSTA
  Eigen3::Eigen
  ${TCL_LIBRARY}
  ${CMAKE_THREAD_LIBS_INIT}
  ${CUDD_LIB}
  )
```

This link propagates to OpenROAD when OpenSTA is built as part of OpenROAD, however when `USE_SYSTEM_OPENSTA` is set and OpenSTA is used as a `.a`, this link does not propagate and the STA symbols are not found.

### Expected Behavior

Something similar to the code above should also be in OpenROAD's CMake manifests -- I understand `USE_SYSTEM_OPENSTA` is something of hassle and I thank you for continuing to support it (it cuts down build times tremendously with caching build systems like Nix.)

### Environment

```shell
-- The CXX compiler identification is Clang 16.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/c1ib07kj7baid3s3wf5m98hcpbvgs8mk-clang-wrapper-16.0.6/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15240-gc95d6a16d
-- System name: Darwin
-- Compiler: Clang 16.0.6
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found OpenSTA: /nix/store/m5hcixmg1lx51hgbyl5f20x6394xhlc9-opensta/lib/libOpenSTA.a
-- Using OPENSTA_INCLUDE_DIR=/usr/include/sta
-- Using ABC_INCLUDE_DIR=/usr/include/abc
-- TCL library: /nix/store/7cpj5zdlz28y6ipg2h4cw6qagvqycm0i-tcl-8.6.13/lib/libtcl.dylib
-- TCL header: /nix/store/7cpj5zdlz28y6ipg2h4cw6qagvqycm0i-tcl-8.6.13/include/tcl.h
-- TCL readline library: /nix/store/0cwgrwkw5kfshdz7bkqjmhm9clas9fqs-tclreadline-2.3.8/lib/libtclreadline.dylib
-- TCL readline header: /nix/store/0cwgrwkw5kfshdz7bkqjmhm9clas9fqs-tclreadline-2.3.8/include
-- Found SWIG: /nix/store/q6jgc1rpam7xqrylwq6gxq2f3arcmhvs-swig-4.0.2/bin/swig (found suitable version "4.0.2", minimum required is "4.0")
-- Found Boost: /nix/store/67hsz87y3shaq2qsql57w5gg8wih8g2m-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0")
-- boost: 1.83.0
-- Found Python3: /nix/store/0hhzvkw889bsybhqxy12ky4jx6a95p2d-python3-3.11.9/include/python3.11 (found version "3.11.9") found components: Development Development.Module Development.Embed
-- Found ZLIB: /nix/store/5g9k8zyjv029pw87xf95pci414r8pw8j-zlib-1.3.1/lib/libz.dylib (found version "1.3.1")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- spdlog: 1.13.0
[... 93 lines cut ...]
      sta::Power::findInputInternalPower(sta::Pin const*, sta::LibertyPort*, sta::Instance const*, sta::LibertyCell*, sta::PwrActivity&, float, sta::Corner const*, sta::PowerResult&) in libOpenSTA.a(Power.cc.o)
      sta::Power::findInputDuty(sta::Instance const*, sta::FuncExpr*, sta::InternalPower*) in libOpenSTA.a(Power.cc.o)
      sta::Bdd::funcBdd(sta::FuncExpr const*) in libOpenSTA.a(Bdd.cc.o)
      sta::Bdd::ensureNode(sta::LibertyPort const*) in libOpenSTA.a(Bdd.cc.o)
      ...
  "_Cudd_T", referenced from:
      sta::Power::evalBddDuty(DdNode*, sta::Instance const*) in libOpenSTA.a(Power.cc.o)
  "_Cudd_bddAnd", referenced from:
      sta::Bdd::funcBdd(sta::FuncExpr const*) in libOpenSTA.a(Bdd.cc.o)
  "_Cudd_bddBooleanDiff", referenced from:
      sta::Power::evalBddActivity(DdNode*, sta::Instance const*) in libOpenSTA.a(Power.cc.o)
      sta::Power::evalDiffDuty(sta::FuncExpr*, sta::LibertyPort*, sta::Instance const*) in libOpenSTA.a(Power.cc.o)
      sta::Power::findInputInternalPower(sta::Pin const*, sta::LibertyPort*, sta::Instance const*, sta::LibertyCell*, sta::PwrActivity&, float, sta::Corner const*, sta::PowerResult&) in libOpenSTA.a(Power.cc.o)
      sta::Power::findInputDuty(sta::Instance const*, sta::FuncExpr*, sta::InternalPower*) in libOpenSTA.a(Power.cc.o)
  "_Cudd_bddCompose", referenced from:
      sta::Sim::funcBddSim(sta::FuncExpr const*, sta::Instance const*) in libOpenSTA.a(Sim.cc.o)
  "_Cudd_bddIthVar", referenced from:
      sta::Bdd::ensureNode(sta::LibertyPort const*) in libOpenSTA.a(Bdd.cc.o)
  "_Cudd_bddOr", referenced from:
      sta::Bdd::funcBdd(sta::FuncExpr const*) in libOpenSTA.a(Bdd.cc.o)
  "_Cudd_bddXor", referenced from:
      sta::Bdd::funcBdd(sta::FuncExpr const*) in libOpenSTA.a(Bdd.cc.o)
ld: symbol(s) not found for architecture arm64
clang-16: error: linker command failed with exit code 1 (use -v to see invocation)
```

jfshorter, 2024-08-18:
There at least seems to be existing prescient / intended permission to do so, think it is doable although I'm not much of a CMake magician.

My guess would be to change this out for a shared include for the library detection, similar to patterns used for other libraries, and add the missing conditions for explicit link-only case.

Replace in the original block(s) in `src/sta/CMakeList.txt` , swap out to use the include.
Then special cases in `src/CMakeList.txt` by adding the include in either the first `else()` or add a new else case around line 400.

maliberty, 2024-08-19:
@donn how urgent is this for you?  It might take a bit to get opensta to factor out the cudd cmake code.  If you need it urgently then I could just copy it over to OR for now.

donn, 2024-08-19:
Not urgent! I already have a dirty patch I can apply during the build process, so take all the time you need to do it right.