The-OpenROAD-Project/OpenROAD #5605: [ERROR PSM-0069] Check connectivity failed on VDD.

oharboe, 2024-08-16:
### Describe the bug

untar https://drive.google.com/file/d/1TT3HmBQ8lHD7UG01vn4RCcwHJOuLCVGb/view?usp=drive_link

```
$ ./run-me-mock-array-asap7-base.sh 
OpenROAD v2.0-15240-gc95d6a16d 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Deleted 0 routing obstructions
[INFO RCX-0431] Defined process_corner X with ext_model_index 0
[INFO RCX-0029] Defined extraction corner X
[INFO RCX-0435] Reading extraction model file /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7/rcx_patterns.rules ...
[INFO RCX-0436] RC segment generation MockArray (max_merge_res 50.0) ...
[INFO RCX-0040] Final 48321 rc segments
[INFO RCX-0439] Coupling Cap extraction MockArray ...
[INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
[INFO RCX-0442] 9% of 86704 wires extracted
[INFO RCX-0442] 14% of 86704 wires extracted
[INFO RCX-0442] 19% of 86704 wires extracted
[INFO RCX-0442] 28% of 86704 wires extracted
[INFO RCX-0442] 33% of 86704 wires extracted
[INFO RCX-0442] 39% of 86704 wires extracted
[deleted]
[WARNING PSM-0038] Unconnected node on net VDD at location (12.228um, 250.290um), layer: M4.
[WARNING PSM-0038] Unconnected node on net VDD at location (12.228um, 250.830um), layer: M4.
[WARNING PSM-0038] Unconnected node on net VDD at location (12.228um, 251.370um), layer: M4.
[WARNING PSM-0038] message limit reached, this message will no longer print
[ERROR PSM-0069] Check connectivity failed on VDD.
Error: final_report.tcl, 61 PSM-0069
openroad> 
```

I made the element generate_abstract standalone test-case too:

```
make DESIGN_CONFIG=designs/asap7/mock-array/Element/config.mk generate_abstract_issue
```

untar https://drive.google.com/file/d/1bWQELFQ6t2y0l5UlZL2x4pwShwii3geF/view?usp=sharing

### Expected Behavior

No error

### Environment

```shell
OpenROAD v2.0-15240-gc95d6a16d
```

### To Reproduce

See above

oharboe, 2024-08-16:
@maliberty @vvbandeira Odd... how did this error make it into master? Would have thought CI would have caught it?

oharboe, 2024-08-16:
@jeffng-or FYI

vvbandeira, 2024-08-16:
> @maliberty @vvbandeira Odd... how did this error make it into master? Would have thought CI would have caught it?

We do not run all ORFS flow tests for each OR code change, so designs might break when using the head of OR.
The "blessed" version of OR in ORFS is the version committed in the submodule, which usually lags behind a couple of days.

We do run CI for the latest OR code, and CI did catch:
https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/4374/pipeline-console/

oharboe, 2024-08-17:
@gadfort @maliberty I have fixed the access for the files above, also added instructions to reproduce locally using make.

oharboe, 2024-08-17:
@gadfort Please confirm that you can reproduce using the instructions and files(should all be fixed now) in this github issue.

gadfort, 2024-08-17:
@oharboe I was able to access and identify the issue.

closed 2024-08-17 by maliberty