The-OpenROAD-Project/OpenROAD #5607: DRT creates redundant arc net when connection to reconfiguration cell

titan73, 2024-08-17:
### Describe the bug

The cell is reconfigurable. Its base layers are the same and the function can be changed with different metal2 connections.

Interesting area are show in orange.

DRT tries to connected to the Q pins of the cell :

[image]

Since the Q pins are connected together in the cell with metal2, DRT connects to the metal2 shape

[image]

But the router create a useless arc wich causes a min hole violation.

[image]

This happens oftern. I have around 100 violations like this in my desgin.

I can try to make a testcase if needed.

### Expected Behavior

No redundant arc

### Environment

```shell
penROAD v2.0-15230-g858535dfe
```

### To Reproduce

No testcase yet

maliberty, 2024-08-17:
Its hard to say much without a test case.

titan73, 2024-08-26:
I reproduce the issue in a test case:

[arc.zip](https://github.com/user-attachments/files/16747119/arc.zip)

titan73, 2024-10-22:
Any news on this issue?

closed 2024-10-26 by osamahammad21

titan73, 2024-10-26:
Why closing this issue without any explanation? A test case is provided.

maliberty, 2024-10-26:
The issue was fixed and you can see the link to the fixing commit above.

titan73, 2024-10-26:
Oups sorry. Didn't pay attention. :s I'll test the fix. Thanks for fixing.