The-OpenROAD-Project/OpenROAD #5608: Installing OpenRoad in an Intel chip Mac fails to compile lemon-graph

aviteri, 2024-08-17:
### Describe the bug

Following the installation guide from https://openroad.readthedocs.io/en/latest/user/Build.html, when executing ./etc/DependencyInstaller.sh the  intallation stops with the following error:

==> Fetching the-openroad-project/lemon-graph/lemon-graph
==> Downloading https://github.com/The-OpenROAD-Project/lemon-graph/archive/refs/tags/1.3.1.tar.gz
Already downloaded: /Users/aviteri/Library/Caches/Homebrew/downloads/8cd49562d55d98cf2eee71a84193ad5d47cbb5285d6682f8252b3e921b1dec2e--lemon-graph-1.3.1.tar.gz
==> Installing lemon-graph from the-openroad-project/lemon-graph
==> cmake -B build .
==> cmake --build build -j --target install
Last 15 lines from /Users/aviteri/Library/Logs/Homebrew/lemon-graph/02.cmake:
In file included from /usr/local/include/boost/multiprecision/traits/is_variable_precision.hpp:9:
In file included from /usr/local/include/boost/multiprecision/detail/number_base.hpp:29:
In file included from /usr/local/include/boost/lexical_cast.hpp:30:
/usr/local/include/boost/lexical_cast/detail/buffer_view.hpp:27:34: error: expected '(' for function-style cast or type construction
        return buffer_view<CharT>{begin, end};
               ~~~~~~~~~~~~~~~~~~^
/usr/local/include/boost/lexical_cast/detail/buffer_view.hpp:31:33: error: expected '(' for function-style cast or type construction
        return buffer_view<char>{
               ~~~~~~~~~~~~~~~~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
1 warning generated.
1 warning and 20 errors generated.
gmake[2]: *** [lemon/CMakeFiles/lemon.dir/build.make:188: lemon/CMakeFiles/lemon.dir/soplex.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:336: lemon/CMakeFiles/lemon.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/The-OpenROAD-Project/homebrew-lemon-graph/issues

### Expected Behavior

I would expect the installation to continue.

### Environment

```shell
Git commit: 984248ab3387df80e10b839b96353bfe8e8a28eb
kernel: Darwin 23.5.0
os: 		macOS 		14.5
cmake version 3.30.2
CMake Error at /usr/local/Cellar/cmake/3.30.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find SWIG (missing: SWIG_EXECUTABLE SWIG_DIR) (Required is at
  least version "4.0")
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.30.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.30.2/share/cmake/Modules/FindSWIG.cmake:184 (find_package_handle_standard_args)
  src/CMakeLists.txt:169 (find_package)

-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-15242-g984248ab3
-- System name: Darwin
-- Compiler: AppleClang 15.0.0.15000309
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/local/Frameworks/Python.framework/Versions/3.12/bin/python3.12 (found version "3.12.5") found components: Interpreter
[... 74 lines cut ...]
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17 - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/local/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Run in an xterminal (MacOS Sonoma 14.5)
`git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
`
`cd OpenROAD`

`./etc/DependencyInstaller.sh`

### Additional Context

I could not report this issue as suggested :
`If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/The-OpenROAD-Project/homebrew-lemon-graph/issues'
There were no "issue" link alternative as the one here.

maliberty, 2024-08-17:
@eder-matheus have you seen this?

eder-matheus, 2024-08-17:
I think I installed lemon by hand a long time ago. I never had issues with it after that. I can try to add it to the dependency installer

jfshorter, 2024-08-18:
Seen similar but different compiler errors due to older compilers or versions of libboost using constructs that were too new for the compiler to cope with.

* Case 1: Compiler constructor support
  * The more modern forms I specifically remember for GCC G++ had dotted member names was broken: `typeName {.member1 = varVal1, .member2 = varVal2...}`
  * Unfortunately this is a little different to this case but might just happen to look like a similar failure mechanism.
* Case 2: Decoupled header chains
  * Often libraries have to make a choice about polluting compile units or making it easy to work with
  * Changes in their requirements to be locally included in the compile unit (i.e: referenced within a C/CPP file) between older versions of libboost and a preceding failure caused a later syntax error.

This is something you can work around, just requires tackling down the header definitions and experimenting with each in isolation to A/B/C test.

stefanottili, 2024-08-23:
The dependency installer works fine on M1 Sonoma.
It uses homebrew to install boost 1.86.0, so boost headers should be at /opt/homebrew/opt/boost not /usr/local/boost

donn, 2024-08-26:
It is most definitely `/usr/local/include/boost` on an Intel Mac.