The-OpenROAD-Project/OpenROAD #5614: DRT Refactor ta

osamahammad21, 2024-08-17:
### Description

TA; Track Assignment, module is responsible for assigning tracks for the given global route guides. The tracks are needed later when we split the design into workers to get the connecting pins of each net that is divided across workers. The refactor should address the 2 main steps in track assignment:
- [ ] initialization (FlexTA_init.cpp)
- [ ] assignment (FlexTA_assign.cpp)

Other parts of the code are easier to understand so we mainly need to document and follow the google code style c++
- [ ] FlexTA.cpp
- [ ] FlexTA_end.cpp
- [ ] FlexTA_graphics.cpp
- [ ] FlexTA_rq.cpp

maliberty, 2025-01-29:
@osamahammad21 do you feel this is done with @bnmfw work?

alokkumardalei-wq, 2026-03-04:
Thanks @maliberty for assigning this to me I would definitely try my best to contribute to resolve this issue , if I face any issues regarding the codebase or the issue itself while working on it or if any doubts I have I would definitely reach out to you by commenting in this thread.