The-OpenROAD-Project/OpenROAD #5628: Question Regarding the Timing Report in the GUI

ehdml, 2024-08-21:
### Describe the bug

I am writing to ask a question regarding the use of the Timing Report and report_checks commands in the OpenROAD GUI.

I found that some paths appearing in the GUI's Timing Report do not get saved when using the command 

**report_checks -slack_max 0 -group_count 1000000 > path.txt.**

I am using the UART of ASAP7.

### Expected Behavior

Saving paths with negative slack from the timing paths. 

I found that some paths appearing in the GUI's Timing Report do not get saved when using the command 

**report_checks -slack_max 0 -group_count 1000000 > path.txt.**

Could you please explain the differences between the Timing Report in the GUI and the report_checks command that I used?

### Environment

```shell
I do not have the necessary permissions to execute the ./etc/Env.sh command.
```

### To Reproduce

[files.zip](https://github.com/user-attachments/files/16688216/files.zip)
openraod -gui
In GUI 
read_lef "./asap7_tech_1x_201209.lef"
read_lef "./asap7sc7p5t_28_R_1x_220121a.lef"
read_liberty "./asap7sc7p5t_INVBUF_RVT_TT_nldm_220122.lib"
read_liberty "./asap7sc7p5t_SIMPLE_RVT_TT_nldm_211120.lib"
read_liberty "./asap7sc7p5t_OA_RVT_TT_nldm_211120.lib"
read_liberty "./asap7sc7p5t_SEQ_RVT_TT_nldm_220123.lib"
read_liberty "./asap7sc7p5t_AO_RVT_TT_nldm_211120.lib"
read_def "./uart_gp.def"
read_sdc "./constraint.sdc"
source "./setRC.tcl"
estimate_parasitics -placement

### Screenshots

[image]
[image]

### Additional Context

In GUI Timing Report show -45.339 path but at path.txt doesn't exist -45.339 path

oharboe, 2024-08-21:
report_checks reports the worst slack path, so -46.095, rounded to -46.10, which you observe. You have highlighted the second worst slack. If you want report_checks to report that, you need to specify that report_checks should report more than just the worst path. Try adding "-group_count 2" to report_checks and you should see the top two lines from the GUI...

ehdml, 2024-08-21:
I used the following command to save all paths with negative slack. 

**report_checks -slack_max 0 -group_count 1000000 > path.txt.**

oharboe, 2024-08-21:
Gotit. Surprising... Could it be that report_checks has 1 path per start/end point and that the GUI is configured to list more paths per endpoint? Click on "Settings" and paste a screenshot...

maliberty, 2026-03-24:
This issue has been automatically marked as stale because it has not had recent activity. It will be closed in 21 days if no further activity occurs. Remove the Stale label or comment to keep it open.