The-OpenROAD-Project/OpenROAD #5632: CTS tech characterization enhancement

arthurjolo, 2024-08-22:
### Description

For the design ihp-sg13g2/riscvi CTS is not inserting buffers in some levels, the image shows the net _clknet_0_clk_ where the level 0 clock buffer (_clkbuf_0_clk_) is driving all the buffers on the level 4.

The tech characterization step determines that for levels 1-3 no buffer is the best option. This makes me believe that there might be an error or some improvements to be done on tech characterization.

arthurjolo, 2024-08-22:
[image]

maliberty, 2024-08-22:
How many sinks are there?

arthurjolo, 2024-08-22:
There are 1056 sinks in this design.

maliberty, 2024-08-22:
That seems like a lot.  How large are the sink clusters?

arthurjolo, 2024-08-22:
The sinks are clustered in groups of 8 to respect the max_fanout limit.